teslamotors/vehicle-command #215: Energy endpoints no longer returning total_pack_energy and energy_left

tomhollander, 2024-03-21:
Today the /products and /energy_sites/{id}/live_status endpoints stopped returning values for `total_pack_energy` and `energy_left`. These are documented in the API and developers should expect them to be returned in every response. Can you please ensure the API is kept stable and any breaking changes are versioned and developers given warning before any deprecations. In the case of these specific values, they are important for owners to understand how their Powerwalls are performing so they should continue to be available.

ThomasAlxDmy, 2025-01-23:
Hey Tom!

The team is telling me: we've deprecated these fields in flavor of `percentage_charged` instead for easier management. We'll look into how we can best communicate these type of changes moving forward!

thefireblade, 2025-01-23:
Going to close old issue for now

closed 2025-01-23 by thefireblade