teslamotors/vehicle-command #216: BLE adding public key not working - Request sent but nothing happens

ManuelStalder, 2024-03-22:
Hi, I'm trying to establish a connection to my Tesla Model 3 with my raspberry pi 4 with Bluetooth. At the end, I would like to wake up my tesla with the raspberry pi, which is not possible with the tesla app or api because we don't have mobile signal in the underground garage (When the car sleeps, it sleeps forever).

I've tryed the tesla-control tool, but with no lock. Beforehand I've created a key with 
> openssl ecparam -genkey -name prime256v1 -noout > private.pem
and 
> openssl ec -in private.pem -pubout > public.pem
Then I tryed several variations with both the "original" tesla-control and also from another work.
> ~/go/bin/tesla-control -vin <MyVin> -ble -debug add-key-request public.pem owner cloud_key

I tried also with parameter variation like driver instead of owner, nfc_card instead of cloud_key etc. I'm always ending with this message:
> Sent add-key request to <MyVin>. Confirm by tapping NFC card on center console.

But nothing happens. Whether the car is locked or unlocked, I do not see any message that I should confirm the new key and also tapping the NFC card to the pillar or middle console does not help.

From studying the source code and some comments I think that it should be possible to add this way my public key and that some actions would then be possible with Bluetooth without having other stuff like a fleet API app registration or without having to use the innoficial owner api (so far my understanding). The raspberry pi should be in range - I tryied from several meters away until I ended up having it inside of the car. The car and the raspberry pi have WLAN internet access, if that matters.

Any help is appreciated.
[image]

erikgieseler, 2024-03-22:
If the message "Confirm by tapping NFC card on center console" appears in the terminal, you must place the Telsa card on the center console. A message only appears when the card is on the center console, no message is displayed before this.

itsMeDavidV, 2024-03-23:
Yeah so can confirm that this is occurring on my end as well. 

Essentially when you goto whitelist a key, the keycard animation no longer shows on the Tesla screen, but the confirm alert does. This confused me so I can only imagine Tesla customers who are not familiar with this process.

ManuelStalder, 2024-03-24:
@erikgieseler Thank you for your response. I was now able to add my key. I was expecting a prompt on the screen. I also placed the NFC card on the center console, but not in the right place. Anyways, after putting the NFC card near the arm rest it worked and the message on the screen asked whether I want to add the new key.
To in the end I fetched to code freshly and used this sequence:
> mkdir test1
> cd test1
> git clone https://github.com/teslamotors/vehicle-command.git
> cd vehicle-command/
> go get ./...
> go build ./...
> go install ./...
> PATH=$PATH:~/go/bin
> tesla-keygen -key-file priv.pem create > pub.pem
> tesla-control -key-file priv.pem -ble -vin <MyVin> add-key-request pub.pem owner cloud_key

Here one has to put/tap the NFC card onto the middle console near the armrest. Then the screen should prompt you If you would like to add the key.

Now I'm able to send all these command (they are working), however the wake does not really wake the car:
> tesla-control -key-file priv.pem -ble -vin <MyVin> ping
> tesla-control -key-file priv.pem -ble -vin <MyVin> honk
> tesla-control -key-file priv.pem -ble -vin <MyVin> charging-set-amps 8
> tesla-control -key-file priv.pem -ble -vin <MyVin> wake
> tesla-control -key-file priv.pem -ble -vin <MyVin> charging-start
> tesla-control -key-file priv.pem -ble -vin <MyVin> charging-stop
> tesla-control -key-file priv.pem -ble -vin <MyVin> flash-lights
> tesla-control -key-file priv.pem -ble -vin <MyVin> charging-set-limit 82
> tesla-control -key-file priv.pem -ble -vin <MyVin> list-keys

Interestingly I am able to wake up my car with the first, slightly different code where I also generated the key with openssl as state in the first post instead of with tesla-keygen. I'm investigating this further.

Shall I close this topic as the original question is resolved (Sorry I'm not really active on github)?

ManuelStalder, 2024-03-24:
Found the solution, why waking up the car was not working. Somewhere noted in a german tesla forum, that one has to add the domain, which worked for me:
> tesla-control -key-file priv.pem -ble -vin MyVin -domain vcsec wake

erikgieseler, 2024-03-24:
You need to use the following command for waking the car up when asleep:
`tesla-control -key-file private_key.pem -ble -domain vcsec -vin LRW* wake`

ManuelStalder, 2024-03-24:
@erikgieseler You're right and you also wrote in the first place, but I overread it. Thank you anyways!

ManuelStalder, 2024-03-25:
Just to give more information for future users. After sending this command:

> tesla-control -key-file priv.pem -ble -vin MyVin add-key-request pub.pem owner cloud_key

The tool then gives you a message like this:

> Sent add-key request to MyVin. Confirm by tapping NFC card on center console.

You must tap your NFC card on the middle console as written earlier, near the armrest. This looks this way at my Model 3:

[image]

After tapping the NFC card, a confirmation message should pop up at the screen. Mine is obviously in german, but it ask more or less whether you would like to allow the new key:

[image]

closed 2024-03-25 by ManuelStalder