teslamotors/vehicle-command #221: How can I obtain the battery capacity of the vehicle, measured in kWh?

idonkeyliu, 2024-03-28:
can only retrieve the battery level from the interface.

thefireblade, 2025-01-23:
That value can be manually calculated given a vehicle's configuration and the state of charge

closed 2025-01-23 by thefireblade

stx, 2025-01-23:
@thefireblade How? The battery level and capacity don't match. 0% battery does not mean 0% capacity, 50% battery does not mean 50% capacity, etc.

thefireblade, 2025-01-23:
@stx If you’re looking for a way to calculate it, it would be using the charging history endpoint and calculate the state of charge % difference from start to end with the amount of kwh added. Of course, the battery % is not precise since the car computer tends to recalibrate from time to time meaning that the state of charge measurement is variable.

As of right now, there is no way to get the exact kwh capacity of the vehicle.