teslamotors/vehicle-command #224: vehicle_data endpoint charge_energy_added sometimes goes backwards

purcell-lab, 2024-03-30:
I sometimes see the charge_energy_added response going backwards when charging a vehicle.

Is this expected behaviour?

jbanyer, 2024-03-30:
I have seen this happen when the vehicle is not charging: the charge_energy_added value can flap up and down slightly.

I haven't observed it when charging, but we don't really use the value so we wouldn't notice if it went backwards.

purcell-lab, 2024-04-02:
Here's an example from the streaming entity during charging  going backwards.

ACChargingEnergyIn: 3.291, 3.270, 3.417

```log
data: {"data":{"ACChargingPower":"5.100","BMSState":"Charge","ACChargingEnergyIn":"3.144"},"createdAt":"2024-04-02T02:21:21.071986716Z","vin":"5YJ3F7EC5LF698201"}
data: {"data":{"ACChargingPower":"5.100","BMSState":"Charge","ACChargingEnergyIn":"3.270"},"createdAt":"2024-04-02T02:21:51.073026767Z","vin":"5YJ3F7EC5LF698201"}
data: {"data":{"ACChargingPower":"5.100","BMSState":"Charge","ACChargingEnergyIn":"3.291"},"createdAt":"2024-04-02T02:22:21.071057565Z","vin":"5YJ3F7EC5LF698201"}
data: {"data":{"ACChargingPower":"5.100","BMSState":"Charge","ACChargingEnergyIn":"3.270"},"createdAt":"2024-04-02T02:22:51.073411747Z","vin":"5YJ3F7EC5LF698201"}
data: {"data":{"ACChargingPower":"5.100","BMSState":"Charge","ACChargingEnergyIn":"3.417"},"createdAt":"2024-04-02T02:23:21.070641762Z","vin":"5YJ3F7EC5LF698201"}
data: {"data":{"ACChargingPower":"5.100","BMSState":"Charge","ACChargingEnergyIn":"3.417"},"createdAt":"2024-04-02T02:23:51.074006926Z","vin":"5YJ3F7EC5LF698201"}
data: {"data":{"ACChargingPower":"5.100","BMSState":"Charge","ACChargingEnergyIn":"3.417"},"createdAt":"2024-04-02T02:24:21.071414160Z","vin":"5YJ3F7EC5LF698201"}
data: {"data":{"ACChargingPower":"5.100","BMSState":"Charge","ACChargingEnergyIn":"3.417"},"createdAt":"2024-04-02T02:24:51.071301393Z","vin":"5YJ3F7EC5LF698201"}
data: {"data":{"ACChargingPower":"5.100","BMSState":"Charge","ACChargingEnergyIn":"3.542"},"createdAt":"2024-04-02T02:25:21.071561073Z","vin":"5YJ3F7EC5LF698201"}
```