teslamotors/vehicle-command #226: Incorrect/unclear HTTP response codes

llamafilm, 2024-03-31:
The Fleet API documentation describes a variety of HTTP response codes which are helpful for building error handling logic.  But I sometimes get a 500 response when a different code would be more appropriate.

> 500: {"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}

This should be 408.

> 500: {"response":null,"error":"vehicle rejected request: your public key has not been paired with the vehicle","error_description":""}

This seems closer to 401 or 412.

> 500: {"response":null,"error":"context deadline exceeded","error_description":""}

OK, 500 makes sense here.

jaggerc, 2024-05-22:
> Hello, 
> 
> 
> 
> do you see a pattern behind this error? 
> 
> 
> 
> `500: {"response":null,"error":"context deadline exceeded","error_description":""}
> 
> `
> 
> 
> 
> I get a lot of that one..
> 
> 

I've been seeing this lately and a reboot of the car seems to clear it up immediately.

vbarrier, 2024-05-24:
Reboot the car is not an option for us :'(  (as rental company)

Hello @sethterashima any thoughts on this?

thefireblade, 2025-01-22:
The status code for `vehicle unavailable: vehicle is offline or asleep` is currently 503, though I agree that this should be 408

I also agree that this error `vehicle rejected request: your public key has not been paired with the vehicle` should be closer to a 403 or 412.

I've filed an internal ticket to address this

closed 2026-09-25 by sethterashima