teslamotors/vehicle-command #231: All commands through the proxy return "error": "not_found"

ClaireDMT, 2024-04-12:
Hello, I would appreciate some help with the proxy implementation.

I set up my third-party client account, registered, received an OAuth token, and my public key on my website under `.well-known`. I can interact from my website with cars through the Rest API. No problems there.

But I have problems making the proxy work (locally for now).

- I have downloaded the SDK repo locally
- Use `go run cmd/tesla-keygen/main.go` to generate private/public key and also the tls certificate for my local linux machine
- I can launch the prox server

But all commands I send through the proxy , even the ones which don't need proxy (vehicle_data ) return the same error: `{"response":null,"error":"not_found","error_description":"","txid":"cece28b72385b06f9f3bf2c88bb0cee8"}`.

**Here is the logs from the proxy**

`2024-04-12T11:08:16+02:00 [info ] Received POST request for /api/1/vehicles/<REDACTED/command/charge_start
2024-04-12T11:08:16+02:00 [debug] Executing charge_start on <REDACTED>
2024-04-12T11:08:16+02:00 [info ] Starting dispatcher service...
2024-04-12T11:08:16+02:00 [info ] Requesting session info from DOMAIN_INFOTAINMENT
2024-04-12T11:08:16+02:00 [info ] Requesting session info from DOMAIN_VEHICLE_SECURITY
2024-04-12T11:08:16+02:00 [debug] Sending request to https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/<REDACTED/signed_command: {"routable_message":"MgIIAzoSEhBbYvzQ1kJRqQVheC5yKB/JmgMQwLa96XsSOe/GugwAIGIF83JDCkEExOSqN8kz9ZKymAxgrKL92ezX51gq85DFdZO8vf6HnghnuhL5Qnw3KCpqAzfVab4jbQ4NncGzpOKgf2MhM0uJFw=="}
2024-04-12T11:08:16+02:00 [debug] Sending request to https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/<REDACTED/signed_command: {"routable_message":"MgIIAjoSEhD2Wty9cYlokKIKoXUNh6xbmgMQxDgxXezTDY7fHwtnD7SVeHJDCkEExOSqN8kz9ZKymAxgrKL92ezX51gq85DFdZO8vf6HnghnuhL5Qnw3KCpqAzfVab4jbQ4NncGzpOKgf2MhM0uJFw=="}
2024-04-12T11:08:16+02:00 [debug] Server returned 404: Not Found: {"response":null,"error":"not_found","error_description":"","txid":"cece28b72385b06f9f3bf2c88bb0cee8"}
2024-04-12T11:08:16+02:00 [warn ] [c438315decd30d8edf1f0b670fb49578] Terminal transmission error: {"response":null,"error":"not_found","error_description":"","txid":"cece28b72385b06f9f3bf2c88bb0cee8"}
2024-04-12T11:08:16+02:00 [error] Returning error Not Found
2024-04-12T11:08:16+02:00 [debug] [c0b6bde97b1239efc6ba0c00206205f3] Retrying transmission after error: Post "https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/<REDACTED/signed_command": context canceled`

sethterashima, 2024-04-12:
Can you run `tesla-control get api/1/vehicles` and verify that you see the expected VIN?

Are you using the [partner OAUTH token](https://developer.tesla.com/docs/fleet-api#partner-authentication-token) (allows you to manage your developer account) instead of the [customer-provided OAUTH token](https://developer.tesla.com/docs/fleet-api#third-party-token) (allows you to send commands to that customer's cars)?

ClaireDMT, 2024-04-15:
Hi,

So when I use the partner OAuth token I get:
`{"response":null,"error":"not_found","error_description":"","txid":"b322db093a55b07cbcb5108502e23748"}`

When using the customer-provided OAuth token, I get this error:
`{"response":null,"error":"vehicle rejected request: your public key has not been paired with the vehicle","error_description":""}`

even if I just paired it:
[image]

We generated a public key for our app without the SQK, following the documentation for the API and it is stored in .well-known folder. So we are able to use the API.

Thanks, I am going in circles here...

ClaireDMT, 2024-04-17:
Thank you, I solved it this morning, I was creating a new key with tesla-keygen and not migrating the key I already generated myself into it.

closed 2024-04-17 by ClaireDMT