teslamotors/vehicle-command #232: Feature request: Allow "charging_manager" role to use "wake" and "charge-port-open" via BLE

erikgieseler, 2024-04-13:
I have just tried out the "charging_manager" role: 
The commands: `charging-start`, `charging-stop` and `charging-set-amps` work without any problems via BLE.

Unfortunately, the following commands do not work via BLE: 
`wake` and `charge-port-open`. 

The following error messages are displayed:
`wake: Failed to execute command: vcsec could not execute command: genericError:GENERICERROR_UNAUTHORIZED`
`charge-port-open: Failed to execute command: MESSAGEFAULT_ERROR_INSUFFICIENT_PRIVILEGES`

Are these two problems fixable ?

_Originally posted by @erikgieseler in https://github.com/teslamotors/vehicle-command/issues/230#issuecomment-2050077059_

sethterashima, 2024-04-15:
Thanks for the report. This will require a vehicle firmware update to fix.

erikgieseler, 2024-05-28:
Is there any news on this?

stumbaumr, 2024-06-16:
Is there any news on this?

sethterashima, 2024-06-20:
> Is there any news on this?

Support for `wake` was merged into a coming firmware release. Support for open/closing the charge port by Charging Manager keys makes sense intuitively, but I'm having second thoughts on that because the role was initially intended for remote management. What's the use case where it would be helpful to have a Charging Manager open/close the ports? Presumably there's a human nearby with a more privileged role in cases where this would be useful.

stumbaumr, 2024-06-21:
> What's the use case where it would be helpful to have a Charging Manager open/close the ports?

You come home, park the car and as Home Assistant knows the car is home and since the saying is "Steht er, dann lädt er" it opens the charge port as soon as you go in park.
Opening the garage door -> Home assistant unlocks the charge port...

erikgieseler, 2024-06-21:
> Support for `wake` was merged into a coming firmware release.
>
> Support for open/closing the charge port by Charging Manager keys makes sense intuitively, but I'm having second thoughts on that [...]

Of course, you can also directly select a role with the highest rights and send the commands via BLE. 
However, for security reasons alone, you should select a role with minimal rights. 
The Charging Manager role would be well suited for this.

pkuehnel, 2024-07-10:
> Support for `wake` was merged into a coming firmware release.

Great news. Any idea of the Version number when this will be possible?

rcasta74, 2024-09-14:
> Support for `wake` was merged into a coming firmware release.

 I got recently the update to `2024.26.7` but I still get `GENERICERROR_UNAUTHORIZED` using a key with role `charging_manager`. What's the firmware version that bring this feature?
Thanks.

thefireblade, 2025-01-23:
closing this old issue, please re-open a new issue if this is still occurring

closed 2025-01-23 by thefireblade