teslamotors/vehicle-command #233: http mode for Google Cloud Run?

RonnyWinkler, 2024-04-13:
I tried to set up the proxy as container in Google Cloud Run environment. 
The proxy only allows https. 
The Google Load Balancer uses https on client side (with Googles app.run certificates), but uses http behind this balancer to forward request to the container.

Is there any chance to use http?
Or does anyone know how to configure cloud container environments like GCR to use https on backend?

sethterashima, 2024-04-15:
Non-technical hobbyists will inevitably shoot themselves in the foot if we make it easy to use `tesla-http-proxy` without TLS. So we won't be doing that.

However, `tesla-http-proxy` is a fairly thin CLI on top of the [Proxy package](https://pkg.go.dev/github.com/k-katsuda/vehicle-command/pkg/proxy), and so users should be able to create a HTTP-only client without needing to fork the repository.

closed 2024-04-15 by sethterashima

liorboyango, 2024-12-02:
> I tried to set up the proxy as container in Google Cloud Run environment. The proxy only allows https. The Google Load Balancer uses https on client side (with Googles app.run certificates), but [...]

Facing the same problem... Were you able to find a solution?

RonnyWinkler, 2024-12-02:
> Facing the same problem... Were you able to find a solution?

I forked the repository and added http/http handling based on used port.
If you are using port 80 or 8080, http is used.
Then I used my repository to create CloudRun container and set port 80 in the Goggle container config.

https://github.com/RonnyWinkler/teslamotors-vehicle-command/blob/main/cmd%2Ftesla-http-proxy%2Fmain.go#L113

liorboyango, 2024-12-02:
@RonnyWinkler that worked, thank you!

liorboyango, 2024-12-03:
> I forked the repository and added http/http handling based on used port. If you are using port 80 or 8080, http is used. Then I used my repository to create CloudRun container and set port 80 in the [...]

Any luck with fleet-telemetry?