teslamotors/vehicle-command #238: Server returned 429: Too Many Requests: Retry later (hidden by tesla-command)

fabianhu, 2024-05-11:
Hi,
I try to solar-charge my Tesla 
- in fact it was working beautifully for months now. 
But now I get the above error.
It seems, that if the weather is a little cloudy, the amount of 50 requests gets eaten away really fast. 
Is there any plan to lift this limitation?

But now for the software part:
tesla-command hides the fact of "Too Many Requests: Retry later" and only gives "Error: context deadline exceeded".
While doing so, it produces 10 retries to the server (where I don't know, if these are also counted by the server)

Please advise, how the retry can be avoided and how the result of 429 can be obtained without parsing the -debug output.
Is there any way to know, how many requests are left?

Many thanks!

PS: using the tesla-command for my python [implementation](https://github.com/fabianhu/tesla_api).

sethterashima, 2024-05-13:
I'll look into the 429.

Have you looked into sending solar charging commands over BLE to avoid command limits?

fabianhu, 2024-05-13:
Thanks, I eyeballed the BLE, but my Pi would have to move near the garage. Wifi is great there, but the Pi is at the other side wired to the heating system. Or I would have to make a relay-Pi.  An interface over Wifi directly would also be nice - but this is car stuff and out of scope here.

sreuter, 2024-05-14:
Tesla just recently started to enforce their (pretty low) rate limits. I left another comment regarding the situation here: https://github.com/alandtse/tesla/issues/961 .. It likely created big trouble for most of their consumers tbh, as they also didn't communicate enforcement beforehand.

fabianhu, 2024-05-17:
Mentioning #52 here, It would be nice to get some data over the BLE interface. I can see the car by pinging it over BLE, but I don't know connection or SOC. Any help is appreciated. 
PS: I just got my relay-Pi.

fabianhu, 2024-05-18:
Did another test today via the API: The 429 appears on the 17th command sent to the vehicle this day. 
Does the tesla-command send multiple requests with one call and they are all counted?

-> Looking at the debug output: Yes it does send 3! 
1. Requesting session info from DOMAIN_INFOTAINMENT
2. Requesting session info from DOMAIN_VEHICLE_SECURITY
3. and the actual command 
== 3
50/3 = 16.6 -> the 17th command does not get through.

Is there a way to send a command "one shot"?

Unfortunately the tesla-command does not provide any output about the "retry after" value in minutes, even in Debug.

I would still opt for a WiFi interface locally. Just use the same stack and auth like the fleet API on a local port.

PR #243 (2024-05-20): Fast fail 429
# Description

Fast fail on 429 errors to give clients better control over retry policies.

Document TESLA_CACHE_FILE environment variable, which may help clients avoid 429s to begin with.

Fixes #238 

## Type of change

Please select all options that apply to this change:

- [ ] New feature (non-breaking change which adds functionality)
- [x] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] Documentation update

# Checklist:

Confirm you have completed the following st [... cut]
closed 2024-05-20 by sethterashima