teslamotors/vehicle-command #239: Is location needed for VehicleControlWindowAction?

itsMeDavidV, 2024-05-15:
Is location needed as a parameter here?

```
message VehicleControlWindowAction {
    LatLong location = 1;
    oneof action {
        Void unknown = 2;
        Void vent = 3;
        Void close = 4;
    }
}
```

I noticed you can send window control commands without the `location` parameter populated and the vehicle will execute the command successfully. I also noticed the vehicle-command SDK also does not populate the `location` parameter.

So going forward I'd like to have a little clarity on this: is location data ever needed for window control and if so when?

If not, will the proto be amended?

sethterashima, 2024-05-17:
The location parameter is not required on vehicles that require newer protocol. Will remove from the protobuf.

itsMeDavidV, 2024-05-17:
> not required on vehicles that require newer protocol

Ah so location parameter would still be required for older vehicles that don't use the protocol?

closed 2024-05-17 by sethterashima