teslamotors/vehicle-command #242: Getting report from users that boombox command is returning 403

itsMeDavidV, 2024-05-17:
As the title says. Our client sends the remote_boombox command without using the protocol since remote_boombox is [not yet supported](https://github.com/teslamotors/vehicle-command/blob/7bfff282d1a9f3149626f8843c8be4363162bc40/pkg/proxy/command.go#L50), however lately some users are reporting that they are now receiving 403s only on this command which indicates that use of the protocol is [now required](https://developer.tesla.com/docs/fleet-api#response-codes) for remote_boombox.

Will the protos and Vehicle-Command now support remote_boombox (or in the near future)?

itsMeDavidV, 2024-05-17:
Update, can confirm that remote_boombox looks to be longer supported as an unsigned command as all attempts now return 403

sethterashima, 2024-05-22:
Duplicate of #109

closed 2024-05-22 by sethterashima