teslamotors/vehicle-command #249: tokens with incorrect ou_code (usually eu users) are not handled for wake up

jbanyer, 2024-05-28:
It appears that the Tesla auth API regularly issues tokens with an incorrect `ou_code`. In particular, Tesla accounts in the EU region will sometimes be issued a token with `ou_code=NA`.

These tokens also include both the NA and EU API URLs in the `aud` value.

The command proxy uses the `ou_code` to [select the appropriate URL](https://github.com/teslamotors/vehicle-command/blob/0d9d955d338d02d9dd14230cca0455535e60d183/pkg/account/account.go#L107) from the `aud` list. This results in selecting the incorrect URL, which then leads to http 421 errors.

The command proxy has a [workaround](https://github.com/teslamotors/vehicle-command/blob/0d9d955d338d02d9dd14230cca0455535e60d183/pkg/connector/inet/inet.go#L150) to detect the 421 error and update the stored URL to the correct one as returned in the error reply.

However, this "detect and adjust" workaround does not work for wake up commands sent to the wrong URL, as they don't seem to return anything. This means that if a vehicle is asleep before any commands have been sent, or the proxy is restarted, it is impossible to wake the vehicle.

jbanyer, 2025-01-09:
FYI this is still occurring as of Jan 2025. A small number of our EU users have an access token with `ou_code = NA`

thefireblade, 2025-01-24:
This is the same issue as https://github.com/teslamotors/vehicle-command/issues/131. Will work on a solution to address this, but it's not going to look pretty either...

closed 2025-01-24 by thefireblade