teslamotors/vehicle-command #255: API returns "Too many requests" (429)

Urkman, 2024-06-07:
Since about two hours the vehicle_data endpoint returns "Too many requests" for my car :(
The response is:
"Retry in 37829 seconds"

What? In next try ion over 10 hours?

Is this a bug? I currently get data every 60 seconds and this was working two hours ago...

There are no announcements on the Tesla developer site:
https://developer.tesla.com/docs/fleet-api#announcements-amp-api-changelog

Urkman, 2024-06-07:
So, it looks like Tesla activated the rate limits without any notice :(

rileymd88, 2024-06-07:
The rate limits have been published for a while but they've recently just
started enforcing them. According to some other GitHub comments new
plans/tiers should hopefully be available in Q3.

andig, 2024-06-07:
Problem with the rate limit- if they have indeed been activated- is that there's no "other" plan than just Discovery.

Urkman, 2024-06-07:
And one big problem here is, that the vehicle_data endpoint still does not return cached data :(

Docu:
`Makes a live call to the vehicle. This may return cached data if the vehicle is offline.`

Urkman, 2024-06-07:
So, that's the data we can work with...
[image]

Urkman, 2024-06-08:
And that's not even true... 
When I limit is fine and I get an answer, I don't get these Headers... Joke?

Urkman, 2024-06-08:
@sethterashima 
Can you say something about this?

Urkman, 2024-06-08:
Is there no one at Tesla able to develop a working API?

iainwhyte, 2024-06-09:
> So, that's the data we can work with... <img alt="Bildschirmfoto 2024-06-07 um 20 07 59" width="329" src="https://private-user-images.githubusercontent.com/94297/337750014-d144bbc9-45ee-4fd4-b805-d85c743496fa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc5MTM1NTIsIm5iZiI6MTcxNzkxMzI1MiwicGF0aCI6Ii85NDI5Ny8zMzc3NTAwMTQtZDE0NGJiYzktNDVlZS00ZmQ0LWI4MDUtZDg1Yzc0MzQ5NmZhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA5VDA2MDczMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThhYjQwMGI4ZDg4YTQ5OGI5NmZjZjg2MmNkNGE5NGYzYTI0OTExMDE1Njg3Nzc1ZGFmZjIzNjhiOTQ5MDllYTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5foVpqqHM1zLadfIQuM9K3CLl57GYAU0kTRkJoKxTp4">

> So, that's the data we can work with... <img alt="Bildschirmfoto 2024-06-07 um 20 07 59" width="329" src="https://private-user-images.githubusercontent.com/94297/337750014-d144bbc9-45ee-4fd4-b805-d85c743496fa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc5MTM2MDMsIm5iZiI6MTcxNzkxMzMwMywicGF0aCI6Ii85NDI5Ny8zMzc3NTAwMTQtZDE0NGJiYzktNDVlZS00ZmQ0LWI4MDUtZDg1Yzc0MzQ5NmZhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjA5VDA2MDgyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWY4YzI2MjkzMjNmMjU3NGY1ZjIyY2ExZDdmOTQwMjM3OWVjOWFjYzU5ZmUxYjU2YjFjMTUwOWUxNDg0MmEyNzYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8vSuEW8pRurXfnvtJmY4Rm5at9I7teZ17dGpbKB-M40">

Is there a way to see the limits with the Tesla-control reference app, or only in the headers of a direct call ?

Urkman, 2024-06-09:
When you get the data from every call, then you can work with them.
But as get them only when it is already to late, this doesn't help you in any case :(

haroldboom, 2024-06-13:
Why enforce an API rate limit when there is no option for people to pay for more calls???

I get Tesla is trying to save money but why not charge for more calls then? Dont enforce a rate limit on customers when there is no other option to get more calls.

Urkman, 2024-06-15:
Does anybody know the exact time, when the rate limit count is reset?
So I could calculate the best time between requests, as the headers are still not send :(

iainwhyte, 2024-06-16:
> Why enforce an API rate limit when there is no option for people to pay for more calls???
> 
> I get Tesla is trying to save money but why not charge for more calls then? Dont enforce a rate limit on customers when there is no other option to get more calls.
> 
> Also the 24 hour ban is way to harsh, it should be 300 seconds before another call can be made.
> 
> The most annoying thing for myself and others is why do we need to do cloud based api calls anyway? I have a Tesla wall charger 3, let me control the charging through that like every other manufacturer. I specifically purchased a Tesla after researching which brand is best for tinkering with open source as I wanted to use HA, Teslamate, EVCC etc and it seems like every week we are chasing our tail trying to debug an API change (always with no notice).

While I don't condone the very ordinary way this was turned on without warning, and no documentation, no communication from Tesla, I do get why - imagine across 2-3M vehicles the number of API calls for commands being made will be enormous. They need it to be efficient, and one way to do that is to limit the API calls. 50 a day seems paltry, so 5 per hour or something, and enforced hourly seems more sane.

That said, I was not being very efficient with my calls. If I wanted to turn on charging to 80% at 32 amps, I would send a wake, then a start charging, then a set limit then a set amps - 4 calls, often 2 or 3 of them return with an 'already set' which I just didn't bother about. Now, I use a call to the current status (not counted in the command API limit), and only send what's needed. Don't wake if already awake, don't set limit if limit is already at 80, don't change amps etc unless a change is needed. The result is my overnight charging needs now fit in the 50 limit without too much hassle. 

That leaves solar charging. It's winter here, so not really in use, but pulling logs from a solar adjusted charge session from a few months back is hundreds of adjustments in ~4 hours. 

I've gone to the trouble of delving into the BLE control. Will adjust my tools to prefer that, and use cloud as a backup. That was the limit is not a problem with the added bonus of on-site commands that don't require internet at all.

haroldboom, 2024-06-16:
> While I don't condone the very ordinary way this was turned on without warning, and no documentation, no communication from Tesla, I do get why - imagine across 2-3M vehicles the number of API calls [...]

Yeah I agree there were people like myself running 30 second api calls to adjust charging however, I am happy to pay to do this and we weren’t given the option which is my major gripe. I had to spend money on a new pi, a Bluetooth dongle and about 2 days trying to get them to all work. That money could have gone to an api bank similar to ChatGPT where you pay per call.

abreingan, 2024-06-16:
> Now, I use a call to the current status (not counted in the command API limit) - iainwhyte. 
I have been using TeslaMate to retrieve data and the Fleet API (via the home assistant Tesla Custom Integration - alandtse/tesla) for commands to allow solar charging. 
This worked well, but TeslaMate seemed to be rate limited for a while recently so that may stop working shortly. 
I was therefore interested to hear iainwhyte saying status calls don't count towards the 50. Does anyone know if that call is the same one used to Poll in the Tesla Custom Integration for Home Assistant as there was some mention that setting the polling too often was tipping the command count.

iainwhyte, 2024-06-16:
> Now, I use a call to the current status (not counted in the command API limit) - iainwhyte.
> I have been using TeslaMate to retrieve data and the Fleet API (via the home assistant Tesla Custom [...]

[image]
The data calls are also rate limited, but have a higher amount (200 vs 50).

https://developer.tesla.com/docs/fleet-api#api-best-practices

abreingan, 2024-06-16:
Thanks Ian.

sethterashima, 2024-06-21:
This scope of this issue tracker is limited to the Golang code in this repository. Please reach out to the Fleet API support team: https://developer.tesla.com/docs/fleet-api#help-and-support.

closed 2024-06-21 by sethterashima