teslamotors/vehicle-command #256: charging-set-amps behaves different via BLE than via FleetAPI

int2001, 2024-06-12:
When setting/calling `charging-set-amps 1` via FleetAPI the vehicle charges with exactly 1A.
Doing the same via BLE, it charges with 2A. I wasn't able to figure out how to limit the current to only 1A.

Bug? Feature?

jbanyer, 2024-06-12:
I have observed that Tesla vehicles don't report the charging amps correctly when below 5A. Perhaps that is related?

sethterashima, 2024-06-12:
I believe @jbanyer is correct; it looks like vehicles don't allow setting the charging rate to below 5A. I see 5A in the charging tab when I send a BLE command to charge at 2A. What model are you using, @int2001? Where are you seeing the 1A (Fleet API) and 2A (BLE) being reported?

int2001, 2024-06-13:
Using Model 3 here.
Saw the 2A at my measuring device connected between Grid and Wallbox. 
Let me try to reproduce it. 

BTW: Yesterday i was wondering about the fact that it was really charging with 1A. But that was the first time. 

All tries before ended in:
- Setting car to 1A (via BLE)
- Measuring-device was never below 2A (watched it for about 2h)
- iOS-App showed 5A (when doing the same via Fleet-API it shows correctly 1A)

kirichkov, 2024-06-14:
From what I've read the minimum limit in the app is 5A because below 5A losses become significant portion of the drawn energy, i.e. the MCU should be powered on and the car to be awake. This will account for those 200 Watts (1 Amp) extra your measuring device is reporting. The Model 3, when awake, is drawing about 300 Watts from the low voltage battery and when connected to the grid, it'd prefer to power itself from the grid.

LubosD, 2024-06-16:
You're probably charging from low voltage (120V) and/or a low phase count (1), right?

Because with 3 phases at 230V, I do get about 1.1 A measured by the wallbox after setting 1 A on the car over BLE.

sethterashima, 2024-06-17:
I haven't been able to reproduce this. BLE/FleetAPI should only affect the transport layer and type of authentication used; the message payload and vehicle-side code are transport agnostic. Additionally, the vehicle-side code appears to enforce a 5A floor. @int2001 could you please provide debug logs from sending the Fleet API command and confirm the current limit that appears in the vehicle UI Charging tab?

andig, 2024-07-09:
Fwiw: in https://github.com/evcc-io/evcc/issues/13007 we've found, that setting vehicle current below 5A (over HTPT) is often ignored on first API call. Once call is repeated or current already is below 5A, it seems to work. So there is definitely something fishy in the stack with lower charge currents.

purcell-lab, 2024-12-13:
Tesla charge on Solar regularly sets charging_amps below 5A.

I too find I need to call the endpoint twice to set a value less than 5A.

With UMC if I set charging_amps to 0A, it will stop all charging which is useful.

With TWC gen 2/gen 3, it doesn't stop charging and sits at 1-2 A