teslamotors/vehicle-command #261: Scan hci0 and hci1

gigich, 2024-06-17:
I have the problem, that my BT-Device is switching between hci0 and hci1. If it gets to hci1, it is no longer under control of this commands for ble communication.

I tried it to rename to hci0 with the following config-files but it does not seem to help:

vi /etc/udev/rules.d/99-bt.rules
KERNEL=="hci1", SUBSYSTEM=="bluetooth", ATTR{address}=="08:BE:AC:43:06:7A", NAME="hci0"

vi /etc/udev/rules.d/10-local.rules
ACTION=="add", SUBSYSTEM=="usb", ATTR{idVendor}=="7392", ATTR{idProduct}=="c611", TEST=="power/control", ATTR{power/control}="on"

SebastianGode, 2024-07-07:
It really should automatically try all available Blutooth devices and not just hci0.
I've got the same issue.

There's a PR open for some BLE improvements: https://github.com/teslamotors/vehicle-command/pull/269
Maybe @sethterashima could also change the behaviour of the client to not only look for hci0 as device.

kirichkov, 2024-07-09:
Unless you're using the `tesla-control` command line utility, I think (haven't tested it just yet) you can select the BlueTooth device using:

```go
	d, err := dev.NewDevice(*device)
	if err != nil {
		log.Fatalf("can't new device : %s", err)
	}
	ble.SetDefaultDevice(d)
```

You need to do this before starting a new connection to the vehicle with `conn, err := ble.NewConnection(ctx, vin)`
I've noticed this in the [go-ble example](https://github.com/go-ble/ble/blob/master/examples/basic/scanner/main.go)