teslamotors/vehicle-command #264: Cannot read vehicles from T4B Admin account

wyvernzu, 2024-06-27:
Hi,

I was working on a third party app registered on Tesla Developer, and by following Tesla API Documentation I was able to pull vehicle and vehicle data from Tesla consumer accounts that have vehicles.

However, when a Tesla business account authenticates via Tesla SSO and generates the access token, I could not see any vehicle through that access token. This Tesla business account has Admin role and it has visability to a list of vehicles on the T4B dashboard, although this account is NOT added as a Driver for any of the vehicles.

My question is that at this moment, if a Tesla business account is NOT listed as a Driver for Tesla vehicles that it 'owns', then is there no way to pull vehicle data through this Tesla business account?

Thank you!

vbarrier, 2024-06-27:
Hi same problem here. Once we add the email as driver, we can see the vehicle but not as the OWNER. 

Also it was written: 
 `*Fleet accounts are excluded from these changes until further notice. Pre-2021 Model S/X are excluded from these changes.` to use new vehicles commands endpoints. It is not the case we receive a lot of errors.

sethterashima, 2024-06-28:
This scope of this issue tracker is limited to the Golang code in this repository. Please reach out to the Fleet API support team: https://developer.tesla.com/docs/fleet-api#help-and-support.

closed 2024-06-28 by sethterashima

themarch, 2024-08-02:
Hi @wyvernzu @vbarrier 

We have the same problem. Every API call with the bearer of the T4B account gives us "unknown user". Were you able to find a solution?