teslamotors/vehicle-command #266: Tesla vehicles require command signing for remote boombox, but its not implemented

Bre77, 2024-06-29:
I am seeing 403 responses for `remote_boombox` now, where it previously has worked, suggesting command signing is now required to run this command, however the http command proxy has not implemented remote boombox.

https://github.com/teslamotors/vehicle-command/blob/cb07d7f792b14a079277d485cd0a5c46efc24ef0/pkg/proxy/command.go#L50

sethterashima, 2024-07-26:
Boombox has various restrictions in different jurisdictions. I've been trying to get approval and disclaimer from our legal team before releasing this code.

itsMeDavidV, 2024-07-27:
> Boombox has various restrictions in different jurisdictions. I've been trying to get approval and a disclaimer from our legal team before releasing this code.

Wouldn't compliance be achieved at the vehicle level though?

Bre77, 2024-07-27:
Media controls are also mostly missing, and now some vehicles require signing for theses too.

liorboyango, 2024-12-25:
Any update here? Waiting for this endpoint for a while now

liorboyango, 2025-02-03:
> Boombox has various restrictions in different jurisdictions. I've been trying to get approval and a disclaimer from our legal team before releasing this code.

Can you please provide an update?

gbhall, 2025-02-25:
Also looking to utilise this endpoint. Are we able to implement this ourselves or will it be blocked serverside? 
It's still in the official documentation: https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-commands#remote-boombox

Bre77, 2025-02-25:
> Also looking to utilise this endpoint. Are we able to implement this ourselves or will it be blocked serverside? 
> It's still in the official documentation: https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-commands#remote-boombox

There is literally no definition for it in the command protocol, which suggests the vehicle firmware doesn't support it. Unless Tesla has intentionally hidden this functionality by removing it from the published protocols, and you could somehow guess the proto layout, it's not possible to implement.

I did raise this with a Tesla developer a few weeks ago as a gap.

gbhall, 2025-05-10:
> > Also looking to utilise this endpoint. Are we able to implement this ourselves or will it be blocked serverside?
> > It's still in the official documentation: https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-commands#remote-boombox
> 
> There is literally no definition for it in the command protocol, which suggests the vehicle firmware doesn't support it. Unless Tesla has intentionally hidden this functionality by removing it from the published protocols (which may actually be true based on [@sethterashima](https://github.com/sethterashima) comment), and you could somehow guess the proto layout, it's not possible to implement.
> 
> I did raise this with a Tesla developer a few weeks ago as a gap.

Tessie has this working for their API, I just tested it and it works:
https://developer.tessie.com/reference/boombox

https://api.tessie.com/{vin}/command/remote_boombox

So not sure how they implemented the feature and it's still working...

Bre77, 2025-05-11:
> Tessie has this working for their API, I just tested it and it works:
> https://developer.tessie.com/reference/boombox
> 
> https://api.tessie.com/{vin}/command/remote_boombox
> 
> So not sure how they implemented the feature and it's still working...

Are you sure your vehicle is command signing required?

@stx are you doing anything magic or undocumented?

gbhall, 2025-05-11:
> Are you sure your vehicle is command signing required?
>
> [@stx](https://github.com/stx) are you doing anything magic or undocumented?

Mine is a new 2025 M3P so I am assuming so...

gbhall, 2025-05-12:
> @Bre77 Yes, we made some effort to make that work on the Tessie API. For the legal reasons Seth mentioned, it's not open-sourced. IIRC, it's due to some countries' laws around the Pedestrian Warning System.

That's awesome you got it working! I'm trying to implement it myself with a fork of this repo. Any chance you could potentially privately share how you accomplished it?

Also the main thing I wanted to know, does it allow you to provide a sound ID other than "fart". For instance an ID of a sound file on the USB etc. 

Also PS boo to lawyers ruining the fun!

liorboyango, 2025-05-12:
> [@Bre77](https://github.com/Bre77) Yes, we made some effort to make that work on the Tessie API. For the legal reasons Seth mentioned, it's not open-sourced. IIRC, it's due to some countries' laws around the Pedestrian Warning System.

I would also like to know how you pulled this off

Bre77, 2025-09-25:
https://github.com/Teslemetry/python-tesla-fleet-api/blob/main/proto/car_server.proto#L86

Bre77, 2025-09-29:
> > https://github.com/Teslemetry/python-tesla-fleet-api/blob/main/proto/car_server.proto#L86
> 
> So you implemented the BT action there? Or the signed http command?

The command protocol is the same over Bluetooth and Fleet API, although certain commands only work on one or the other.

I haven't done a reference Python implementation but I could, and I've got it working in my closed source NodeJS application.

liorboyango, 2025-09-30:
> > > > https://github.com/Teslemetry/python-tesla-fleet-api/blob/main/proto/car_server.proto#L86
> > > 
> > > 
> > > So you implemented the BT action there? Or the signed http command?
> > 
> > 
> > The command protocol is the same over Bluetooth and Fleet API, although certain commands only work on one or the other.
> > I haven't done a reference Python implementation but I could, and I've got it working in my closed source NodeJS application.
> 
> Is it possible to implement the http proxy part in Go in a fork of this repo?

Answering myself - yes it's possible

Cromzinc, 2025-11-12:
> > > > > > https://github.com/Teslemetry/python-tesla-fleet-api/blob/main/proto/car_server.proto#L86
> > > > > 
> > > > > 
> > > > > So you implemented the BT action there? Or the signed http command?
> > > > 
> > > > 
> > > > The command protocol is the same over Bluetooth and Fleet API, although certain commands only work on one or the other.
> > > > I haven't done a reference Python implementation but I could, and I've got it working in my closed source NodeJS application.
> > > 
> > > 
> > > Is it possible to implement the http proxy part in Go in a fork of this repo?
> > 
> > 
> > Answering myself - yes it's possible
> 
> Can you state how please?
> 
> Also is it only able to trigger the "fart" sound or any media file on the USB. What does it actually enable?

I added forked and added the remote boombox. Working great, but no ping. As I mentioned above, appears they have a usage limitation around this. 

I can see why they'd want to limit remote boombox, but odd they allow farts, but not a ping. 

https://github.com/Cromzinc/vehicle-command