teslamotors/vehicle-command #267: Cannot get a refresh token - invalid client ID

Andreas-lang, 2024-07-01:
Dear community,

I am trying to build a charging app, that continuously monitors charging. To do this, I will need to refresh access tokens over time. Unfortunately, I get the following error when trying to acquire a refresh token:

'error': 'invalid_client_id', 'error_description': 'Invalid client_id. Authorize client_id and token client_id need to match.'

**What I do to produce the error**

1. First, I ask the user to authenticate and retrieve the code from my website (callback url). I wrote a short script in Python to make this very reproducible as I debug the error. Here it is:

`base_url = "https://auth.tesla.com/oauth2/v3/authorize"
    params = {
        "response_type": "code",
        "client_id": CLIENT_ID,
        "redirect_uri": REDIRECT_URI,
        "scope": SCOPES,
        "state": STATE,
        "nonce": "somenonce"
    }
    url = f"{base_url}?{urllib.parse.urlencode(params)}"
    return url`

This step work well. Using the returned "code", I get a valid token, and can use it to e.g. send a honk command to the car.

2. Second, I try to get an access token and a refresh token, which Tesla documentation says should be included in the return of the call. Again, I coded it in Python:

`conn = http.client.HTTPSConnection("auth.tesla.com")
    payload = urllib.parse.urlencode({
        "grant_type": "authorization_code",
        "client_id": CLIENT_ID,
        "client_secret": CLIENT_SECRET,
        "code": code,
        "audience": AUDIENCE,
        "redirect_uri": REDIRECT_URI,
        "scope": SCOPES
    })
    headers = {'Content-Type': 'application/x-www-form-urlencoded'}
    print(payload)
    conn.request("POST", "/oauth2/v3/token", payload, headers)
    res = conn.getresponse()
    data = res.read()
    return json.loads(data)`

Unfortunately this is where instead of the token, I get the error.

**What I have tried to troubleshoot it**

1. I made sure that my app is registered. I do get a partner token, and I can register my app.
2. I checked that the client id is transmitted correctly. If I change the client ID I rightfully get an error that no matching client ID was found.

**Where I need your help**

1. Have you implemented this flow and does it work for you?
2. Am I using the right endpoints?
3. Did you encounter this error before and could solve it?

Your help would be much appreciated!

Best,
AKdeveloper

Andreas-lang, 2024-07-01:
I just tried two additional things:

1. I modified the client-ID in the authorisation URL (first call). Predictably, the Tesla website then gives an error that the clientID is invalid. So we can rule out that the first request has an invalid ID.
2. I queried the userinfo endpoint: (https://auth.tesla.com/oauth2/v3/userinfo) with the code as Bearer, and it returns a valid result. So we definitely have a valid token at the right ClientID.

Still puzzled what goes wrong in the second request.

sethterashima, 2024-07-02:
Try setting `grant_type` to `"client_credentials"` and dropping teh `code` value. See [client_credentials](https://developer.tesla.com/docs/fleet-api#partner-authentication-token).

Andreas-lang, 2024-07-03:
Thank you Seth for considering my situation!

I am attempting to refresh a user token. Per instructions on https://developer.tesla.com/docs/fleet-api#third-party-token, I am up to step 3:

> 3. Execute a code exchange call to generate a token. Save the refresh_token to generate tokens in the future. The access_token can be used as the Bearer token in the Authorization header when making API requests.

What you suggested (using grant_type "refresh_token") would be the next step, step 4. 
I tried it it just now directly with my "code" token from step 2, with this code:

`    conn = http.client.HTTPSConnection("auth.tesla.com")
    headers = {'Content-Type': 'application/x-www-form-urlencoded'}
    payload = urllib.parse.urlencode({
        "grant_type": "refresh_token",
        "client_id": CLIENT_ID,
        "client_secret": CLIENT_SECRET,
        "refresh_token": code
    })
    print(payload)
    conn.request("POST", "/oauth2/v3/token", payload, headers)`

And get this error:
{'error': 'server_error', 'error_description': 'Internal server error', 'error_uri': '615922f5-fdad-43b8-918a-048de6c1e045-1719967726530. For more information, visit: https://developer.tesla.com/docs/fleet-api#help-and-support'}

**Note**: This error makes sense to me, because I don't have refresh-token to exchange yet, and I have jumped over step 3.

Andreas-lang, 2024-07-05:
I have done further debugging to isolate what is happening.
1. I decoded the "code" parameter (with https://jwt.io) I extract from the callback in step 2, to see if my token is fully valid. It is.
2. The decoded token has the value "azp", "azp" matches my client-id exactly. So the error message is misleading, the token client_id and the authorize client_id match.

I'll now try revoking access for the app and re-authorising.

Andreas-lang, 2024-07-05:
Further update - I have revoked access to the app and then re-did steps 1 to 3 (give user authorisation URL, authorise, extract code from callback, request authorisation_code) - same error.
`{'error': 'invalid_client_id', 'error_description': 'Invalid client_id. Authorize client_id and token client_id need to match.', 'referenceID': 'f884de38-4541-4565-894e-85091556cebb-1720140632454. For more information, visit: https://developer.tesla.com/docs/fleet-api#help-and-support'}
`

I hope someone knows the trick for this, I am at a loss.

Andreas-lang, 2024-07-07:
Found it. When you get the callback you get multiple things:

- a code URL parameter
- a JSON body with an access_token, a refresh_token and an id_token

For some reason, I had extracted the access_token from the callback. When changing the extract to the URL code parameter it works. The code should look like "NA_...".

closed 2024-07-07 by Andreas-lang