teslamotors/vehicle-command #270: How to wake over BLE

Andreas-lang, 2024-07-05:
I'm trying to wake the car over bluetooth, like this:

`tesla-control -ble wake`

I get one of two errors:

`Failed to execute command: MESSAGEFAULT_ERROR_INSUFFICIENT_PRIVILEGES
`
or

`Error: context deadline exceeded
`

Is it possible to wake over bluetooth? If yes, how did you do it?

System is MacOS 14.5.

erikgieseler, 2024-07-05:
See https://github.com/teslamotors/vehicle-command/issues/248

Andreas-lang, 2024-07-06:
Thank you Erik!

To get around `Error: context deadline exceeded`, per #248, it was important to include the parameter `-domain vcsec` in the call.

To get around the `Failed to execute command: MESSAGEFAULT_ERROR_INSUFFICIENT_PRIVILEGES`, I needed to ensure I was using the right private key. My mistake was that I forgot that the tesla-keygen utility stored the private key in my login keychain, and I needed to set the right environment variable so it could find it again.

As a side-note, how do I export the private key from the Mac OS keychain? I would prefer to have a file with a copy, just in case.

erikgieseler, 2024-07-06:
Probably similar to these instructions https://support.apple.com/sr-rs/guide/keychain-access/kyca35961/mac

Andreas-lang, 2024-07-07:
That's doesn't work for the specific private key. I think for backup, it would be great if I could extract a private key '.pem' file. I'll close this issue for now though, thanks for helping Erik!

closed 2024-07-07 by Andreas-lang