teslamotors/vehicle-command #271: Running HTTP only for traefik reverse proxy

XxTopKillerzZ, 2024-07-06:
Hello,

Is it possible to run the HTTP Proxy in http mode without any server/client certificate?

The example method of launching is tesla-http-proxy -tls-key key.pem -cert cert.pem -port 443
Is it possible to launch like 'tesla-http-proxy -port 80'

Our idea is to run Tesla's HTTP proxy in http and then reverse proxy it trough traefik for certificate generation.

Thank you,
Tomás

mlancer, 2024-07-09:
Right now im using a script I made to extract the cert and key from the traefik acme file once every so many months(after renewal). This was quick and dirty. Also, its probably not the best approach but I'm still in early development for what I'm working on. 
Anyway...If it's of use to you,

```shell
#!/bin/bash

### Use to extract cert and keys from traefik acme file store
###TO RUN enter ./scriptname.sh acme_cf.json tld.com in terminal window

c_red="\033[0;31m"
c_green="\033[0;32m"
c_yellow="\033[0;33m"
c_reset="\033[0m"

success() {
        echo -e "${c_green}$1${c_reset}"
}

warning() {
        echo -e "${c_yellow}$1${c_reset}"
}

error() {
        echo -e "${c_red}$1${c_reset}"
}

die() {
    >&2 echo -e "${c_red}$*${c_reset}"
    exit 1
}
#Incoming File
ACMEJSON=$1
#Domain Name to extract
domain=$2
# Define the output file names
output_cert_file="decoded_certificate.pem"
output_key_file="decoded_key.pem"

if [ ! -f "$ACMEJSON" ]; then
  die "could not find acme json file: $ACMEJSON"
fi

CA=$(jq --arg domain "$domain" -r ' .cloudflaredresolver.Certificates[] | select(.domain.main == $domain) | .certificate' $ACMEJSON)

PRIVKEY=$(jq --arg domain "$domain" -r ' .cloudflaredresolver.Certificates[] | select(.domain.main == $domain) | .key' $ACMEJSON)

# Decode base64 encoded certificate
decoded_cert=$(echo "$CA" | base64 --decode)
decoded_key=$(echo "$PRIVKEY" | base64 --decode)

# Print the decoded certificate (or you can use it in further processing)
#echo "$decoded_cert"
#echo "$decoded_key"

# Save the decoded certificate to a new file
echo "$decoded_cert" > "$output_cert_file"
echo "$decoded_key" > "$output_key_file"
# Inform the file saved location
success "Decoded certificate and key saved to $output_cert_file and $output_key_file"
```

sethterashima, 2024-07-09:
See https://github.com/teslamotors/vehicle-command/issues/233.

closed 2024-07-09 by sethterashima

XxTopKillerzZ, 2024-07-11:
@mlancer So you use the extracted certificate in Tesla HTTP Proxy? Does traefik stay in front of the Proxy or do you only use it for certificate generation?

mlancer, 2024-07-15:
@XxTopKillerzZ Yeah i'm using the extracted certificate for Tesla HTTP proxy. I don't have traefik in front of the proxy, I'm only using it for certificate generation since the same TLD is already behind traefik.

XxTopKillerzZ, 2024-07-15:
Alright. Thats what I ended up doing in my setup a couple days ago.