teslamotors/vehicle-command #272: ble.NewConnection sometimes never returns

wimaha, 2024-07-07:
Hello,

when there are connection issues the function `ble.NewConnection(ctx, vin)` returns with `context deadline exceeded`. That's the expected behavior.
But after a while (> 15 attempts) the function just never returns.

Any clue what could cause this issue?

sethterashima, 2024-07-26:
Could you please include the output of running the command with `-debug` and provide your OS and hardware information?

Jan21493, 2024-08-13:
I had the same or at least a very similar problem. As long as the vehicle is nearby the command tesla-control works well, even if repeated thousands of times over a longer time period. With the new code from a few weeks ago the BLE connection is very stable on my system.

But when the vehicle is driven away, I got the same error `Error: context deadline exceeded` as expected, but after a few times I got `Error: failed to find a BLE device: can't init hci: no devices available: (hci0: can't down device: device or resource busy)`. I also got `Error: failed to find a BLE device: can't init hci: can't create socket: too many open files`. See attached output for details. 

A temporary workaround was to restart the bluetooth service (systemctl restart bluetooth), however this does not last long. Reason for the problem might be a missing 'close' for the underlying socket (socket is created by code from "github.com/go-ble/ble", in ble/linux/hci/socket/socket.go starting lin line 63). I'm neither an expert with Go nor the go-ble package, so I haven't prepared a pull request, but was able to add a few lines of code (see attachment) that look to fix the problem for me. 

Please add that fix (or a better one) .

[error--too-many-open-files.txt](https://github.com/user-attachments/files/16602189/error--too-many-open-files.txt)

wimaha, 2024-08-22:
I think I found the issue. It is in the underlaying bluetooth package.

In the ble.go we have the following code:
https://github.com/teslamotors/vehicle-command/blob/d4c0fedc4162bdb88ec2a35ce80aeb16ac686256/pkg/connector/ble/ble.go#L174C2-L186C3

There the ble package is called via `client, err := ble.Connect(ctx, filter)`.

In the ble package we have the following code in the Connect function (glatt.go):
https://github.com/go-ble/ble/blob/8c5522f543335a80e18fc70e704b104cf3fcc606/gatt.go#L125C1-L149C2

So and here is a bug. When the vehicle can't be found, normally ctx2 fires bevor ctx, so that the function `Scan` returns `context deadline exceeded`. But sometimes (rarely but randomly) ctx fires and cancels ctx2 (see line 131). In this case `Scan` return with `context canceled`. This yields to the line 147 `cln, err := Dial(ctx, (<-ch).Addr())` is executed and ist blocking, because of `(<-ch).Addr()`.

So we need to propose a pull request on the ble package to solve this bug.

wimaha, 2024-08-22:
I've opened a PR on the ble package: https://github.com/go-ble/ble/pull/112

Jan21493, 2024-08-23:
@wimaha: I still have the issue described above when the vehicle can't be found with latest code from 'tesla-control' on a Raspberry PI 3B+. After a while the hci0 interface is busy and requires a reset.

So far I ended up by switching to BLE package from https://github.com/rigado/ble instead of go-ble/ble. The code for BLE has changed a lot and required minor changes in pkg/connector/ble, but now my issue is gone. To me the ble package from rigado looks to be better, but I'm wondering if that code will end up in go-ble some time in the future.

Jan21493, 2024-09-07:
It turned out that the rigado/ble package has the same issue and that your fix solves 'my' problem. I've opened a PR on that package https://github.com/rigado/ble/pull/76

I'm using the '_tesla-control_' utility to track if my car is at home or away via BLE.

The problem is easy to reproduce by using the '_tesla-control_' utility with a non-existing VIN, e.g. 
`tesla-control -ble -vin LRW3E7XXXXXXXXXX5 body-controller-state`
and calling the utility every minute from a script. After some minutes up to some hours the utility hangs which causes the next call to report an error: can't init hci: no devices available: (hci0: can't down device: device or resource busy)

sethterashima, 2024-10-31:
I've looked in vain for a well-maintained Golang BLE library.

If anyone knows of one, please feel free to put up a PR to migrate or create an Issue asking someone else to.

sethterashima, 2025-01-28:
Obviated by https://github.com/teslamotors/vehicle-command/pull/353, which removes the call to `Connect`.

closed 2025-01-28 by sethterashima