teslamotors/vehicle-command #273: Cannot change plan

andig, 2024-07-09:
Having registered early on when only discovery plan was available, I need to upgrade the plan now. Fleet api support does not answer requests in any way. Website does not allow to change plan.

What is the process to upgrade a plan?

thefireblade, 2025-01-22:
For now, you will need to create a new application with a different plan. In the future, may provide a way to update your existing application's plan.

closed 2025-01-22 by thefireblade

jbanyer, 2025-01-23:
@thefireblade does the concept of plan even exist any more?

thefireblade, 2025-01-23:
Fleet API is now usage based, but some accounts may have plans on the business side. I would reach out to Fleet API support for all issues related to accounts, pricing / plans, etc.