teslamotors/vehicle-command #274: Do you need to have your own website hosted to consume Tesla Fleet API?

jenildesai25, 2024-07-14:
I have been banging my head for couple hours to figure out how Tesla Fleet API works, I don't know why Tesla had made it so difficult to use it. Nevertheless, could someone tell me if I need to host a website to receive partner account key?. I don't have website I was looking to play with my car using this commands and was wondering if I can just use Tesla Fleet API without hosting a website. Any help is appreciated, Thank you

sethterashima, 2024-07-15:
You can enter `http://localhost:<port number>` for allowed origin when registering.

However, you will not be able to provide an `https://tesla.com/_ak/<your domain>` link that would allow your users to enroll your virtual key onto their cars.

Andreas-lang, 2024-07-19:
I built a simple website on Vercel to stay flexible. Vercel was its own learning curve, but very useful, fast and stable.

closed 2024-07-22 by sethterashima