teslamotors/vehicle-command #275: BLE body-controller-state returns wrong vehicleSleepStatus

osimmert, 2024-07-15:
The car was parked over night without charger.  With
"sudo tesla-control -domain vcsec -ble -key-file xx.pem -vin xx body-controller-state" I get
{
	"vehicleLockState":  "VEHICLELOCKSTATE_LOCKED",
	"vehicleSleepStatus":  "**_VEHICLE_SLEEP_STATUS_AWAKE_**",
	"userPresence":  "VEHICLE_USER_PRESENCE_NOT_PRESENT"
}
If I start the Tesla-App it tells me that the last update was 14 hours ago, so the car should have slept.
Even if I wait 30 minutes without using the App and not touching the car I get the same result.
It was said that the body-controller-state can be read without waking the car. 
I can't tell if the call accidentally waked the car or if the output message is wrong.
Or the arguments ? Do I need another domain ?

sethterashima, 2024-07-15:
This isn't the expected behavior. Are you able to confirm that the car wasn't accessed by either the Tesla app or a third-party app shortly beforehand?

osimmert, 2024-07-16:
Just rechecked and the first time I got:
{
	"vehicleLockState":  "VEHICLELOCKSTATE_LOCKED",
	"vehicleSleepStatus":  "VEHICLE_SLEEP_STATUS_ASLEEP",
	"userPresence":  "VEHICLE_USER_PRESENCE_NOT_PRESENT"
}
I have to check why the car was awake all the time the other day.
Sorry for the confusion. The issue can be closed.

closed 2024-07-16 by osimmert