teslamotors/vehicle-command #276: This third party in not registered with Tesla

ViNYLeq, 2024-07-15:
Hello!

I was using old API for some time and now realized I cannot control anything from Home Assistant integration perspective. I have followed instructions to create developer account, register application etc. so I have OAuth credentials (client ID and secret), but stuck at the very end.

1. Login to Tesla account - success.
2. Generate token from URL - success but not possible to click "OK" button, need to use "back" button.

[00:59:26] webui:DEBUG: Callback URL: https://tesla.mydomain.com/callback?code=XXXXXXX&state=xxxx&issuer=https%3A%2F%2Fauth.tesla.com%2Foauth2%2Fv3
[00:59:26] webui:DEBUG: code: XXXXXXXX
[00:59:26] urllib3.connectionpool:DEBUG: Starting new HTTPS connection (1): auth.tesla.com:443
[00:59:26] urllib3.connectionpool:DEBUG: https://auth.tesla.com:443 "POST /oauth2/v3/token HTTP/1.1" 200 1395
[00:59:26] webui:WARNING: Obtained refresh token: EU_XXXXXXXXXXXX
[00:59:26] werkzeug:INFO: 172.30.32.2 - - [16/Jul/2024 00:59:26] "GET /callback?callback_url=https://tesla.mydomain.com/callback?code%3DEU_XXXXXXXXXXXXX%26state%3DXXXXXXXX%26issuer%3Dhttps%253A%252F%252Fauth.tesla.com%252Foauth2%252Fv3 HTTP/1.1" 200 -

3. Enroll public key in your vehicle - fails in Tesla app - "Cannot share vehicle. This third party in not registered with Tesla. We can't grant them access at this time."

[01:02:02] werkzeug:INFO: 172.30.32.2 - - [16/Jul/2024 01:02:02] "GET / HTTP/1.1" 200 -

I'm not sure what's wrong, maybe this OK button not functioning, maybe I should do something else to register third party company somewhere. I am the owner of the vehicle, using same account for developer registration and setting up access.

Please help!

ViNYLeq, 2024-07-16:
I will answer myself. I created developer account under other email address, then operation went smooth - don't use same tesla account for driver and developer access.

Marcellion, 2024-07-16:
I have the same problem, unfortunately changing email address from dev account, so that it differs from owners acc. doesn't help for me still same error message: "Cannot share vehicle. This third party in not registered with Tesla. We can't grant them access at this time."

any other suggestions?

swisscheesy, 2024-09-02:
> I have the same problem, unfortunately changing email address from dev account, so that it differs from owners acc. doesn't help for me still same error message: "Cannot share vehicle. This third party in not registered with Tesla. We can't grant them access at this time."
> 
> any other suggestions?

Did you ever find a fix to this?  I tried with both main account and secondary, same issue both times.

Marcellion, 2024-09-03:
Nope, I sort of parked the issue over the summer holidays.

Sw1zero, 2024-10-28:
I'm posting so I see if anything happens. Have the same problem too.

pcoenen, 2024-10-31:
I have the same issue, I'm also using a subdomain as domain. I was wondering if that would be part of the issue? @Sw1zero are you also using a subdomain?

Sw1zero, 2024-12-09:
Yeah ober duckdns.org

GitAashishG, 2024-12-25:
I am running into same issue. I also tried creating a separate email for developer account. From WebUI of Tesla http proxy, When I log into Tesla, do you use developer account or the driver account. I tried both to no success. Can go up to Generate token from URL, but when I go to enroll Key to vehicle - from both app and desktop scanning QR, I get "This third party isn't registered with Tesla. We can't grant them access at this time."

thefireblade, 2025-01-24:
Closing this issue, we're migrating all such account related issues to https://fleetapi-support.tesla.com/

closed 2025-01-24 by thefireblade