teslamotors/vehicle-command #278: How to get information from EncryptedData?

Whirlwind, 2024-07-20:
I found the struct in the protobuf file:
https://github.com/teslamotors/vehicle-command/blob/05bc5dd8d0649b4ccb45a765b9127d06f1050a6f/pkg/protocol/protobuf/car_server.proto#L101-L108

I don't find how to decrypt the `ciphertext`. Anybody can help me?

sethterashima, 2024-07-22:
I believe this is currently used by Sentry Cam Clips and some vehicle data. The former requires push notifications and the latter is in the process of being rearchitected in a way that will break reverse compatibility, so we don't plan to implement the soon-to-be-removed method in the SDK.

closed 2024-07-22 by sethterashima

Whirlwind, 2024-07-23:
> the latter is in the process of being rearchitected in a way that will break reverse compatibility, so we don't plan to implement the soon-to-be-removed method in the SDK.

It said: you are rewrite the `get vehicle data` API, and it will be release soon, right?