teslamotors/vehicle-command #280: How to set the key name by BLE?

Whirlwind, 2024-07-22:
I add the BLE key success. But the key name in the car screen always be the `Unknown Key`.

I could not find the api in the protobuf file to change the key name. Anybody help me?

sethterashima, 2024-07-22:
Key names are changed using a REST API, implemented in the library here: https://github.com/teslamotors/vehicle-command/blob/05bc5dd8d0649b4ccb45a765b9127d06f1050a6f/pkg/account/account.go#L215

closed 2024-07-22 by sethterashima

Whirlwind, 2024-07-22:
Only for the rest API？ is there a protobuf API for  BLE？

sethterashima, 2024-07-22:
Key names are controlled by the backend server so that they can be synchronized across cars, and so they don't support a protobuf API.

If you add your key with an NFC card form factor, you should be able to rename it using the vehicle UI.

Whirlwind, 2024-07-23:
> If you add your key with an NFC card form factor, you should be able to rename it using the vehicle UI.

Yes, I can.

> Key names are controlled by the backend server so that they can be synchronized across cars, and so they don't support a protobuf API.

I am able to rename the key name by the vehicle UI, so the vehicle will send the new name to the backend server, right?
I think that when I rename the key name by the protobuf API, the vehicle  will be able to send the new name to the backend server, too. why is it not?