teslamotors/vehicle-command #281: Where is the `CAR_SEAT_REAR_LEFT_BACK` and `CAR_SEAT_REAR_RIGHT_BACK`

Whirlwind, 2024-07-27:
https://github.com/teslamotors/vehicle-command/blob/05bc5dd8d0649b4ccb45a765b9127d06f1050a6f/pkg/protocol/protobuf/car_server.proto#L140-L166

I known that the `CAR_SEAT_FRONT_XX` is the first row, and the `CAR_SEAT_REAR_XX` is the second row. Right?

I can't imagine which seat the `CAR_SEAT_REAR_LEFT_BACK` and `CAR_SEAT_REAR_RIGHT_BACK` refers to.

NeuWasFlaggedAgain, 2024-07-27:
I'd assume it's the third row in the mx, but there's a `CAR_SEAT_THIRD_ROW_x`. Maybe it's ventilated seats?

sethterashima, 2024-07-29:
It looks like the `CAR_SEAT_REAR_LEFT_BACK` and `CAR_SEAT_REAR_RIGHT_BACK` were created when legacy message types were translated into protobufs. In the legacy protocol, they referred to seat heaters for some pre-2021 Model S vehicles. However, those Model S vehicles don't support the new protocol, and so the protobuf entries are vestigial.

closed 2024-11-01 by sethterashima