teslamotors/vehicle-command #284: What is the `manual_override` in `HvacSetPreconditioningMaxAction`?

Whirlwind, 2024-08-02:
I am looking for the `defrost` API, it seams that the [set_preconditioning_max](https://developer.tesla.com/docs/tesla-fleet-api#set_preconditioning_max) to that.

> ### set_preconditioning_max
> POST /api/1/vehicles/{vehicle_tag}/command/set_preconditioning_max
>
> scopes: vehicle_cmds
>
> ⚠️ Newest vehicles require Tesla Vehicle Command protocol ⚠️
>
> Sets an override for preconditioning — it should default to empty if no override is used.

I check the `Tesla Vehicle Command protocol`, I found the action:
https://github.com/teslamotors/vehicle-command/blob/eaf6ee1fee2bcc2bcbb44dfea921e9fab78ef249/pkg/protocol/protobuf/car_server.proto#L187-L196

What is the `manual_override_mode`? Which mode is the `defrost`?

By the way, I found that there are many `manual_override` fields in the hvac APIs in the protobuf/FleetAPI, What do they mean？

sethterashima, 2024-08-07:
It looks like manual override forces the command to succeed even when battery charge (SoC) is low, the doors are open, or dog mode is enabled. Use with caution.

Whirlwind, 2024-12-01:
@sethterashima Thanks for  your reply. I have other question: The Defrost Mode only control the on/off status, is there a way to control the `FrontDefrosterOn` or `RearDefrosterOn`?
[image]

thefireblade, 2025-01-23:
I don't think we have these controls implemented - this would be a separate feature request issue. Closing issue for now

closed 2025-01-23 by thefireblade