teslamotors/vehicle-command #285: Always get the error 408 (vehicle is offline) when send to DOMAIN_INFOTAINMENT by `signed_command` API

Whirlwind, 2024-08-05:
I send the `session_info_request` cmd to the `DOMAIN_INFOTAINMENT` and `DOMAIN_VEHICLE_SECURITY` by the BLE, everything works.

But I send them by the Fleet API, the `DOMAIN_VEHICLE_SECURITY` works but the `DOMAIN_INFOTAINMENT` always failed:
```
API GET: /api/1/vehicles/xxxxx
{
    response =     {
        "access_type" = OWNER;
        state = online;
        vin = xxxx;
        .....
    };
}

Send >>>>>> VCSEC
<RoutableMessage 0x30314fac0>: {
    to_destination {
      domain: DOMAIN_VEHICLE_SECURITY
    }
    from_destination {
      routing_address: "\213\265I\234\312/\211\263\242\007\341b"
    }
    session_info_request {
      public_key: "\004\312w\000\345I\231\307\007Zlp;\221\201\314?\342m\016\016Is\277\223\305\243\205k$\374\010\215.\203 Yy\267\362\367#\006wj.M\025*\374VE\013k\"\243e\022s{\266.\277\003\245"
    }
}

API POST: /api/1/vehicles/xxxxx/signed_command
{
    response = "Mg4SDIu1SZzKL4mzogfhYjoCCAJ6XBJBBHy889d/Hy9m/wZ3km/KI3mzn8AQyDXMs1MxMdzDo+2Hqx2dqQ7O0juPMd/K7CKUtHX178jkIk9x82qiIic/xkQaEBHWpnnXa7+hL/NXoTy2hOclEwAAADAKaiQyIgogN8a43EIuoWI5c+luOPntI1vNoG3kCUS2ghaCL8oHuO0=";
}

Send >>>>>> Infotainment
<RoutableMessage 0x30314fac0>: {
    to_destination {
      domain: DOMAIN_INFOTAINMENT
    }
    from_destination {
      routing_address: "\030\003\340~./\3301\235pnp"
    }
    session_info_request {
      public_key: "\004\312w\000\345I\231\307\007Zlp;\221\201\314?\342m\016\016Is\277\223\305\243\205k$\374\010\215.\203 Yy\267\362\367#\006wj.M\025*\374VE\013k\"\243e\022s{\266.\277\003\245"
    }
}

API POST: /api/1/vehicles/xxxxx/signed_command (error: Error Domain=TeslaClientError Code=408 "(null)" UserInfo={NSLocalizedFailure=vehicle is offline})
```

I am sure the vehicle is online, I am in the car...

sethterashima, 2024-08-06:
If you're able to reproduce, could you try (1) running `tesla-control` with `-debug` and (2) checking to see if the mobile app works over the Internet?

Whirlwind, 2024-08-06:
I will try it. But the command send to vcsec works by `signed_command`, I can unlock and open the door.

The commands to Infotainment domain failed.

By the way, my region is China, and the tesla host is `https://fleet-api.prd.cn.vn.cloud.tesla.cn`. I am not sure it is a bug for the china region.