teslamotors/vehicle-command #287: BLE: Failed to execute command: car could not execute command: SetChargingAmps failed

erikgieseler, 2024-08-07:
Since some recent update, the charging-set-amps function no longer works via BLE.

Car: Tesla Model 3 LR 2021
Firmware: 2024.26.3.1

This command was executed:
`tesla-control -debug -key-file /home/admin/Tesla/private_key.pem -ble -vin LRW* charging-set-amps 6`

Debug Log:
```
2024-08-07T16:56:52+02:00 [debug] Set session cache file to ''
2024-08-07T16:56:52+02:00 [debug] Set OAuth token name to ''
2024-08-07T16:56:52+02:00 [debug] Set OAuth token file to ''
2024-08-07T16:56:52+02:00 [debug] Set keyring type to ''
2024-08-07T16:56:52+02:00 [debug] Set keyring Debug Logging to 'false'
2024-08-07T16:56:52+02:00 [debug] Client public key: 042a*
2024-08-07T16:56:52+02:00 [debug] Connecting over BLE...
2024-08-07T16:56:52+02:00 [debug] Creating new BLE device
2024-08-07T16:56:52+02:00 [debug] Searching for BLE beacon S*...
2024-08-07T16:56:52+02:00 [debug] Connecting to BLE beacon a4*...
2024-08-07T16:56:52+02:00 [info ] Connected to vehicle BLE
2024-08-07T16:56:52+02:00 [info ] Connecting to car...
2024-08-07T16:56:52+02:00 [info ] Starting dispatcher service...
2024-08-07T16:56:52+02:00 [info ] Securing connection...
2024-08-07T16:56:52+02:00 [info ] Requesting session info from DOMAIN_INFOTAINMENT
2024-08-07T16:56:52+02:00 [info ] Requesting session info from DOMAIN_VEHICLE_SECURITY
2024-08-07T16:56:52+02:00 [debug] TX: 320208023a1212100380c1e0ef246bfc4cb1ce15332ce9079a0310c9564158ca7c49af01d846de0a2f717f72430a41042a5e6dde18e9d23e82b98d7b2fb758b69c60f9b1d62db6c94aac564652aa882c1cd65a3ce505c0b80c858b5520268c88c53154581598b830f8eeb71293bb3097
2024-08-07T16:56:53+02:00 [debug] TX: 320208033a121210e5afe06dbf56e8612a7401c0ab410bd99a03104f65b5182a792f88e741f2bf17e1a96772430a41042a5e6dde18e9d23e82b98d7b2fb758b69c60f9b1d62db6c94aac564652aa882c1cd65a3ce505c0b80c858b5520268c88c53154581598b830f8eeb71293bb3097
2024-08-07T16:56:53+02:00 [debug] RX: 321212100380c1e0ef246bfc4cb1ce15332ce9073a0208027a5e0805124104ba9141ad8beddf96d3a1b26cb29109396cc91b713d75755a2d0bc68d9255616dab616c79e0d0bba48cf36288587b4681a10cb9e53934fde4ae0702d6a04cd0fd1a10418d554227b16e3012912981951d1946256775000030056a2432220a20dd3f0a680c18a7df205972e38bc0692e96142ba49900657b68c1c5eb62ed9fcc920310c9564158ca7c49af01d846de0a2f717f
2024-08-07T16:56:53+02:00 [info ] [c9564158ca7c49af01d846de0a2f717f] Updated session info for DOMAIN_VEHICLE_SECURITY
2024-08-07T16:56:53+02:00 [debug] RX: 320208003a020802521f1a1d12160a14f8ed37c298b23b265cf5e214dc8e88093ac2a0421802220101
2024-08-07T16:56:53+02:00 [debug] [00] Dropping message to DOMAIN_BROADCAST
2024-08-07T16:56:54+02:00 [debug] RX: 32121210e5afe06dbf56e8612a7401c0ab410bd93a0208037a5e08251241042086b185dba7bd603cd8194105877148d87eef10da0dfcfa39f24ecf19c058a48edc6ab455c9ef6f594b36531bd177ac14f06446452a6fad3c6bc5369696b01a1a10df663b3fc0d5072c4078cc21d1e38dcc251feb070030056a2432220a2036e6171d272f43a0d821efbc4f1eb568024eb226875fb052ae3c86f0f62e70f79203104f65b5182a792f88e741f2bf17e1a967
2024-08-07T16:56:54+02:00 [info ] [4f65b5182a792f88e741f2bf17e1a967] Updated session info for DOMAIN_INFOTAINMENT
2024-08-07T16:56:54+02:00 [debug] TX: 320208033a121210e5afe06dbf56e8612a7401c0ab410bd99a0310fd9bb2753c04e6072d89271fe781c4645207c9efc78f965c396a80010a430a41042a5e6dde18e9d23e82b98d7b2fb758b69c60f9b1d62db6c94aac564652aa882c1cd65a3ce505c0b80c858b5520268c88c53154581598b830f8eeb71293bb30972a390a10df663b3fc0d5072c4078cc21d1e38dcc120cdab13cdfb00f82a50f0d988c18262524eb07002a10249d71252e2c32205ff4b00240b1f03e
2024-08-07T16:56:54+02:00 [debug] RX: 320208003a020802521f1a1d12160a14f8ed37c298b23b265cf5e214dc8e88093ac2a0421802220101
2024-08-07T16:56:54+02:00 [debug] [00] Dropping message to DOMAIN_BROADCAST
2024-08-07T16:56:55+02:00 [debug] RX: 32121210e5afe06dbf56e8612a7401c0ab410bd93a020803521e0a1c080112180a165365744368617267696e67416d7073206661696c6564920310fd9bb2753c04e6072d89271fe781c4649a0310347f75700b2bc752018d4ae71da7d348
Failed to execute command: car could not execute command: SetChargingAmps failed
```

sethterashima, 2024-08-09:
Can you confirm that this is reproducible over BLE, but not over the Internet?

The error indicates the infotainment system successfully received and authenticated the command, but could not execute the command. This can happen under normal circumstances if the infotainment system is in the process of booting/waking up and the service responsible for executing the command has not come up yet. But if you can reliably reproduce the error after giving the car time some time to finish waking up, then further investigation might be required.

erikgieseler, 2024-08-09:
Yes, the error persists even after a waiting time of 20 seconds.
I can only test the BLE commands as I don't use the Internet commands because the rate limits are far too low.
However, it makes no difference whether the car is asleep or not. The error also occurs when the car is not asleep.

maart84, 2024-08-13:
I'm having the same error via the web calls. The weird thing is: I can't even change it in the tesla app itself anymore now. As soon as I try to set another Amperage I get a message: commad failed. Only changing it in the car still works and reflects back in the app. Maybe a Tesla bug?
Car: 2019 SR+  with 2024.26.6

zlymeda, 2024-08-15:
I have the same problem that started when I upgraded from `2024.20.9` to `2024.26.3.1`.
When updated to `2024.26.6`, restarting my Tesla once did the trick. I was able to change the amps.

But today, I got updated to `2024.26.7` and can no longer change the amps. I tried restarting the car multiple times, but it did not work.

I tried:
- official Tesla Android app
  - fails via BLE
  - fails via the internet 
- fleet API SDK
  - fails via BLE

Also, I noticed that these commands work:
- lock/unlock car
- turn on/off the climate
- sentry on/off
- start/stop charging
- lock/unlock charge port

These don't work:
- change amps
- open/close windows

I own the Tesla Model 3 LR 2019, if that helps.

erikgieseler, 2024-08-15:
I can also no longer adjust the charging current via the Tesla APP.

zlymeda, 2024-08-16:
interesting. 

- I went for a drive
- restarted the Tesla away from home
- tried the app right there on the parking lot - still getting the error
- came home, plugged it in
- tried to change the amps.. and now it works

doomastor, 2024-08-16:
Hi,
I have also the same problem. As zlymeda, I notice that isssues appeared after upgrading to 2024.26.3. 

My Car is a SR+ 2021.

pkuehnel, 2024-08-16:
Same issue here. Even with the official Tesla App, I cannot change the current anymore. BLE and Fleet API also not working. Changing the current in the car's infotainment screen is working. Issue in my third party app is [here](https://github.com/pkuehnel/TeslaSolarCharger/issues/1420)

coreGreenberet, 2024-08-17:
Same issue over fleet api & tesla app.
Tesla Model Y: 2024.26.7

maart84, 2024-08-17:
Today I installed 2024.26.7 and it started working again.  And in the official app too. 
model 3 2019 rwd.

pkuehnel, 2024-08-23:
Looks like I found a solution. What you need to do: Disconnect the car charger, sit in the car, switch off the Bluetooth of your phone. Make sure that no BLE enabled phone, Key Fob,... is around. Note: BLE has a connection radius of > 20 meters, so best would be to disable Bluetooth of all phones and remove the battery from the key fob. Then restart the car by pressing both scroll wheels, wait for the car to reboot (I waited an additional minute after the navigation screen came back), enable Bluetooth on your device. Now it should work again.

coreGreenberet, 2024-09-02:
> Looks like I found a solution. What you need to do: Disconnect the car charger, sit in the car, switch off the Bluetooth of your phone. Make sure that no BLE enabled phone, Key Fob,... is around. [...]

I didn't believe your solution as I did a normal restart, but removing the fob battery and turning off the BT from my phone did really the trick with the reboot

pkuehnel, 2024-09-02:
Yep, no BLE device beeing around is important. I tried multiple reboots, Software Reinstall,... nothing helped, only a reboot without any BLE device helps. I know of about 15 cars where that helped and not a single one where it did not help.

zhy2020, 2024-09-13:
Hello, can I control the air conditioner through ble? I look at the code as if it is calling the Internet method

Peter919pan, 2024-09-30:
Anyone still experience SetChargingAmps failed in 2024.32.3?
I'm frequently access this API on 2024.20.6, just worry the upgrade hit this bug

coreGreenberet, 2024-09-30:
After doing the restart without BLE devices near the car it did solve my issue (and from many others too). I've updated the car afterwards already on 2024.32.3 and the issue didn't happen again.

I would say it is safe to update, but even if not...just restart the car without a ble device next to it =)

pkuehnel, 2024-09-30:
@coreGreenberet From my experience as maintainer of https://github.com/pkuehnel/TeslaSolarCharger there were some vehicles where disabling all BLE devices after while restarting did not. Some even drove a few miles away from home to make sure there is not an unknown BLE device around, did still not help. @Peter919pan But for All of the updating to 2024.32.3 resolved the issue, currently I am not aware of any car on 2024.32.x that has the issue. But still, there is no confirmation from Tesla that the issue is resolved.

pkuehnel, 2025-01-24:
@sethterashima I think you can close this issue, has been resolved with a firmware update around 2024.32.x

closed 2025-01-24 by sethterashima