teslamotors/vehicle-command #290: Set Scheduled Departure no longer works for 2024.26+

stx, 2024-08-13:
What's the plan to deal with the new scheduled charging and scheduled climate feature in the car and Tesla app with 2024.26+? Will this come to Fleet API or no? As is, the Set Scheduled Departure endpoints no longer work on these vehicles.

jbanyer, 2024-08-16:
In addition, the existing schedule charging fields appear to no longer be set for this new firmware. For example, here are the values for a vehicle (2024.26.7) with a repeating daily charging schedule enabled:

```
    off_peak_charging_enabled: false,
    off_peak_charging_times: 'all_week',
    preconditioning_enabled: false,
    preconditioning_times: 'all_week',
    scheduled_charging_mode: 'Off',
    scheduled_charging_pending: false,
    scheduled_charging_start_time: null,
    scheduled_departure_time: null,
```

This breaks any third-party software that relies on these fields, including the software that I develop :|

Sebazzz, 2024-08-20:
>  As is, the Set Scheduled Departure endpoints no longer work on these vehicles.

Actually, it does work, but you end up with additional schedules. So for every API call a new schedule is added.

sethterashima, 2024-08-22:
> @sethterashima are there plans to add API fields or endpoints to get the new charging schedule?
> 
> I've noticed that the existing fields (scheduled_charging_mode, scheduled_charging_start_time_minutes) are being used with the new schedule, but it's not clear how they work and they obviously can't describe the new schedule model.

@aaronpkahn ?

closed 2024-08-29 by patrickdemers6

themonomers, 2024-09-16:
@sethterashima @patrickdemers6 when I call the [vehicle_data](https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-endpoints#vehicle-data) endpoint and request `charge_schedule_data` to view existing schedules, it seems to return general vehicle information with no details about the car's charge schedules.  Did I miss an update?

The [add_charge_schedule](https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-commands#add-charge-schedule) endpoint seems to work fine.

Thank you for your help!

rmcclell, 2024-12-19:
The endpoint seems to add and remove scheduled departures but cabin overheat protection seems to prevent scheduled departure from triggering the charging at the required time.

PR #305 (2024-08-29): Support new scheduled charging
Adds support for weekly charging schedules.

# Description

Building off of https://github.com/teslamotors/vehicle-command/pull/296

Fixes #290 

## Type of change

Please select all options that apply to this change:

- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] Bug fix (non-breaking change which fixes an issue)
- [ ] Documentation update

# Checklist:

Confirm you have completed the following steps:

- [x] My code follows the style of [... cut]