teslamotors/vehicle-command #309: Can not only change the driver temp when the `sync` is enabled in the vehicle

Whirlwind, 2024-09-04:
We known that we can change the driver/passenger temperature with different celsius, and we also enable the `sync` to keep the same temperature (change the driver temp will also change the passenger's.)

Now, I can not find the Fleet/BLE API to enable/disable the `sync`, and not find the API to get the `sync` state.

If I send a command to change the temperature with different celsius, the vehicle will force change them to same celsius when the `sync` is enabled.

Have any api to control/get the `sync` state?

thefireblade, 2025-01-24:
Not sure what you mean by `sync`, can you elaborate, otherwise I'll close the old issue

Whirlwind, 2025-02-06:
@thefireblade In the vehicle screen, you can adjust the driver temperature, and there is a `sync` button to enable. If I enable the `sync` button, the passenger temperature will be same as the driver temperature.