teslamotors/vehicle-command #310: Climate missing override parameter?

stx, 2024-09-09:
When you start the climate in the Tesla app with a door open it prompts for an override. Presumably using the same manual_override parameter that exists in other commands like Defrost?

That override is present on some commands in the SDK but is missing here: https://github.com/teslamotors/vehicle-command/blob/e8634aa8ac74282ab195edc462a86b1f87f451de/pkg/vehicle/climate.go#L38

zhy2020, 2024-09-13:
Hello, can I control the air conditioner only through ble? I look at the code as if it is calling the Internet method

sethterashima, 2024-09-16:
@stx The Manual Override bool is defined in the protobuf built by the method you linked: https://github.com/teslamotors/vehicle-command/blob/main/pkg/protocol/protobuf/car_server.proto#L142. Be careful about setting it to true; IIRC, it also overrides dog mode and low SoC, and so, as the name suggests, should never be set automatically without explicit manual confirmation.

There's a [pending PR](https://github.com/teslamotors/vehicle-command/pull/308) that will likely conflict with a quick fix, so I'd prefer to wait it gets merged.

@zhy2020 Please keep comments on topic. The [example BLE program](https://github.com/teslamotors/vehicle-command/blob/e8634aa8ac74282ab195edc462a86b1f87f451de/examples/ble/main.go) shows how to accomplish what you're after.

stx, 2024-09-16:
> @stx The Manual Override bool is defined in the protobuf built by the method you linked: https://github.com/teslamotors/vehicle-command/blob/main/pkg/protocol/protobuf/car_server.proto#L142. Be [...]

Makes total sense. Thank you!

thefireblade, 2025-01-24:
closing issue

closed 2025-01-24 by thefireblade