teslamotors/vehicle-command #317: Is there a public commons Vehicle Commands Proxy server?

jverkoey, 2024-09-27:
On https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-commands it mentions "The command must be sent through the Vehicle Commands Proxy to send commands." The use of the word "the" instead of "a" made me wonder if there is a specific Vehicle Commands Proxy server that can be used instead of spinning up an instance of this vehicle-command proxy? E.g. https://commands-proxy.prd.na.vn.cloud.tesla.com/

jverkoey, 2024-09-27:
The reason I'm asking is because I'd prefer not to have to insert a custom server in between a user and their car; a network route that goes from user to car through only official Tesla APIs would be ideal from a privacy and data protection point of view.

programmrz, 2024-09-29:
You'll need to spin up a server to run the proxy. Don't be worried, the command proxy still expects oauth tokens and the like so nothing is really happening on it's own.

jverkoey, 2024-09-29:
I realized after I wrote this that the main reason this proxy is needed is because the private key needs to be used to sign the request, and that private key shouldn't be shipped in client-side applications.

closed 2024-09-29 by jverkoey