teslamotors/vehicle-command #339: How to decryto the `encrypted_data` when request the `media_detail_state`?

Whirlwind, 2024-12-11:
The `GetMediaDetailState` command will get the message:
```
{
    actionStatus {
    }
    vehicleData {
      encrypted_data {
        field_number: media_detail_state
        ciphertext: 94BEC155137DC775BB27A8EC98A2C7434EF583B5735A1E227038FDF7D352F73CDA
        tag: DFFD3C8F0B9D1DAF0FFE088ADD1A58FE728BD6B2B56067571EF9493D
      }
      media_detail_state {
      }
      supports_optional_fields: true
    }
}
```

I don't find how to decryto the `encrypted_data`, could somebody help me?

Whirlwind, 2024-12-15:
@sethterashima I get the `encrypted_data` response from the Model 3 with 2023.38.8, and use the [newer encryption method](https://github.com/teslamotors/vehicle-command/blob/main/pkg/protocol/protocol.md#response-decryption).

sethterashima, 2024-12-16:
Is this over BLE or over the Internet? You might get this format if the car is offline, since location data cached on the server is encrypted using the older method.

Whirlwind, 2025-01-03:
@sethterashima Hi, I upgrade my Model3 to 2024.44.4, and the response is right without the `encrypted_data`. :)

closed 2025-01-03 by Whirlwind