teslamotors/vehicle-command #340: ServerHttp is not called anymore

rawmean, 2024-12-12:
The following function which is supposed to be used for proxy authorization is not called anymore. Do you also see this problem?

```
func (h *CustomHandler) ServeHTTP(w http.ResponseWriter, r *http.Request) {
	// Perform your business logic here

	// Check if the request is authorized
	if isAuthorized(r) {
		// If authorized, invoke the original handler
		h.p.ServeHTTP(w, r)
	} else {
		// If not authorized, return an error response
		http.Error(w, "Unauthorized", http.StatusUnauthorized)
	}
}
```

sethterashima, 2024-12-16:
I'm not seeing that code anywhere in the repository. If `h.p` is a `proxy.Proxy` object that you're adding client authentication to, then double check that you're calling `http.ListenAndServeTLS` on `h` and not `h.p`.

rawmean, 2024-12-16:
Yes, the `if isAuthorized` code is mine (to authorize access).
To make the CustomHandler be called again, I had to change the following change in `cmd/esla-http-proxy/main.go`. 
```
	customHandler := &CustomHandler{p: p}
	log.Error("Server stopped: %s", http.ListenAndServeTLS(addr, httpConfig.certFilename, httpConfig.keyFilename, customHandler))
```

closed 2024-12-16 by rawmean