teslamotors/vehicle-command #342: Scheduled Charging not triggering

rmcclell, 2024-12-19:
The cabin overheat protection seems to preventing the car from sleeping correctly scheduled departure from running correctly. If I turn it off it will trigger charge at correct time. I have a model y and this correctly and my wife has a older model s. This seems to be a regression issue, I've seen it documented on other forums as previously fixed issue. It seems like it is model specific. Her model s is 2018 running the latest update, but it has been happening since September. Hers is the Intel be based MCU. I can get additional car details if that helps trace it down. Spoke with support they just try to sell her a charging port. I develope software for a living and I believe we have isolated it to the specific setting.

thefireblade, 2025-01-24:
Can anyone else reproduce this issue?

rmcclell, 2025-01-24:
The feature that seem to trigger it for my wife was to turn on scheduled
departure and cabin overheat protection her model s is 2018. My vehicle is
a model y 2024 but it works correctly on mine. I think it is a model
specific race condition issue

rmcclell, 2025-04-20:
It appears to still be an issue even with the latest update. My wife had re-enable cabin overheat because it is now getting warmer, and it still preventing the car from sleeping and schedule departure from working correctly.

rmcclell, 2025-04-20:
Can anyone provide some additional steps to gather additional debug information. I have a development background and am willing to help gather the addition debug information you need to isolate and reproduce the issue, but I am not for sure how to gather the correct information or what additional information you need from me. It is very consistent for me when cabin overheat is on the car doesn't wake correct and doesn't charge correctly with scheduled departure, regular charging works fine. I have zero issues on my new Model Y. I can't see it as being hardware issue because normal charging works correctly. It seems something happens when overheat is on the prevents the sleep and wake operations from happening correctly which prevents the scheduler from running on time. It may be a bug that only exists on older Model S's.