teslamotors/vehicle-command #343: Can't remove or rename a key from within the car anymore - security issue.

Jan21493, 2024-12-27:
In summer 2024 I was able to rename and delete a key on the touch screen from within the car from the controls > locks menu, but when I open the same menu now, the symbols to rename or delete a key are grayed out for ALL keys. 

In my opinion this is a security issue, because if either one of the NFC cards oder one of the cloud keys that I've installed on the vehicle to be able to control it via BLE got stolen, I can't revoke / remove the keys anymore.

Tested on my model 3 from 2023 with version 2024.45.32 and 2024.44. 

I know that the issue is caused by the software running in the car and not from this SDK, but I only know this repository where I can create an issue.

sethterashima, 2024-12-27:
Sounds like your car has Guest Mode enabled: https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-commands#guest-mode.

You can turn Guest Mode off using the SDK.

Jan21493, 2024-12-28:
Yes, that was the issue, so it was my fault. I have played around with the new features and I have left guest mode on. Is the state of the guest mode shown somewhere, e.g. via one of the state commands that are available via BLE? 

@sethterashima : thank you for the quick reply!

closed 2024-12-28 by Jan21493