teslamotors/vehicle-command #345: media_control not work both old and new protocol

mrgaolei, 2025-01-05:
old protocol:
```
{
    "response": null,
    "error": "Tesla Vehicle Command Protocol required, please refer to the documentation here: https://developer.tesla.cn/docs/fleet-api/support/announcements#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning",
    "error_description": "",
    "txid": "48ab51b1237b909bfb48304572112368"
}
```

new protocol:
```
{"response":null,"error":"invalid_command","error_description":""}
```

when using new protocol, other command is OK, for example: flash_lights, honk_horn and others.

patrickdemers6, 2025-01-22:
Thanks for the report. Created a PR to address these missing commands.

Lenart12, 2025-01-23:
More missing commands here #188

sethterashima, 2025-08-05:
Fixed in 93dce2b8b1582b33a96bf8b368445023d59001d4

closed 2025-08-05 by sethterashima