teslamotors/vehicle-command #346: Infinite loop in NewConnection

skrul, 2025-01-12:
Hey folks, I've been working on some automated charging using the [TeslaBleHttpProxy](https://github.com/wimaha/TeslaBleHttpProxy) project which uses `vehicle-command` to communicate with the car. From time to time I've seen failures where the process runs amok, taking up a lot of memory and a lot of cpu. I enabled debug logging (`ble.SetDebugLog()`) and noticed the following three log lines getting printed out in a tight loop when things go sideways:

```
Reusing existing BLE device
Searching for BLE beacon S26d2d43805843xxxx...
BLE connection attempt failed: failed to find BLE beacon for 5YJ3E1Exxxxxxxxxx (S26d2d43805843xxxx): can't scan: received scan response 60:10:98:c1:06:98 with no associated Advertising Data packet
```

I believe the `for` loop in `NewConnection` ([here](https://github.com/teslamotors/vehicle-command/blob/main/pkg/connector/ble/ble.go#L130)) is looping and never exiting. Seems like this can happen if `err` is non-nil but `ctx.Err()` is nil.

More logs:

```
{"log":"2025/01/12 00:04:19 INFO connecting to Vehicle ...\n","stream":"stderr","time":"2025-01-11T23:04:19.910961044Z"}
{"log":"2025/01/12 00:04:19 DEBU trying connecting to vehicle attempt=1\n","stream":"stderr","time":"2025-01-11T23:04:19.910996239Z"}
{"log":"2025/01/12 00:04:19 DEBU connecting to vehicle (A)...\n","stream":"stderr","time":"2025-01-11T23:04:19.911008109Z"}
{"log":"2025-01-12T00:04:19+01:00 [debug] Reusing existing BLE device\n","stream":"stderr","time":"2025-01-11T23:04:19.911082279Z"}
{"log":"2025-01-12T00:04:19+01:00 [debug] Searching for BLE beacon S26d2d43805843xxxx...\n","stream":"stderr","time":"2025-01-11T23:04:19.911091327Z"}
{"log":"2025-01-12T00:04:20+01:00 [warn ] BLE connection attempt failed: failed to find BLE beacon for 5YJ3E1Exxxxxxxxxx (S26d2d43805843xxxx): can't dial: can't dial: receiv
ed scan response 60:10:98:c1:06:98 with no associated Advertising Data packet\n","stream":"stderr","time":"2025-01-11T23:04:20.050659487Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Reusing existing BLE device\n","stream":"stderr","time":"2025-01-11T23:04:20.050689462Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Searching for BLE beacon S26d2d43805843xxxx...\n","stream":"stderr","time":"2025-01-11T23:04:20.050695201Z"}
{"log":"2025-01-12T00:04:20+01:00 [warn ] BLE connection attempt failed: failed to find BLE beacon for 5YJ3E1Exxxxxxxxxx (S26d2d43805843xxxx): can't scan: received scan response 60:10:98:c1:06:98 with no associated Advertising Data packet\n","stream":"stderr","time":"2025-01-11T23:04:20.05074225Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Reusing existing BLE device\n","stream":"stderr","time":"2025-01-11T23:04:20.050762181Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Searching for BLE beacon S26d2d43805843xxxx...\n","stream":"stderr","time":"2025-01-11T23:04:20.050768482Z"}
{"log":"2025-01-12T00:04:20+01:00 [warn ] BLE connection attempt failed: failed to find BLE beacon for 5YJ3E1Exxxxxxxxxx (S26d2d43805843xxxx): can't scan: received scan response 60:10:98:c1:06:98 with no associated Advertising Data packet\n","stream":"stderr","time":"2025-01-11T23:04:20.050828Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Reusing existing BLE device\n","stream":"stderr","time":"2025-01-11T23:04:20.050836273Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Searching for BLE beacon S26d2d43805843xxxx...\n","stream":"stderr","time":"2025-01-11T23:04:20.050841228Z"}
{"log":"2025-01-12T00:04:20+01:00 [warn ] BLE connection attempt failed: failed to find BLE beacon for 5YJ3E1Exxxxxxxxxx (S26d2d43805843xxxx): can't scan: received scan response 60:10:98:c1:06:98 with no associated Advertising Data packet\n","stream":"stderr","time":"2025-01-11T23:04:20.050845921Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Reusing existing BLE device\n","stream":"stderr","time":"2025-01-11T23:04:20.050850743Z"}
{"log":"2025-01-12T00:04:20+01:00 [debug] Searching for BLE beacon S26d2d43805843xxxx...\n","stream":"stderr","time":"2025-01-11T23:04:20.050855109Z"}
{"log":"2025-01-12T00:04:20+01:00 [warn ] BLE connection attempt failed: failed to find BLE beacon for 5YJ3E1Exxxxxxxxxx (S26d2d43805843xxxx): can't scan: received scan response 60:10:98:c1:06:98 with no associated Advertising Data packet\n","stream":"stderr","time":"2025-01-11T23:04:20.05090224Z"}
... same three lines repeat forever
```
Thanks for having a look at this!

Jan21493, 2025-01-12:
Do you have seen this: https://github.com/teslamotors/vehicle-command/issues/272#issuecomment-2305506012 ?

Using the BLE package with the PR#112 fixed a similar issue for me. I'm polling my vehicle every 30 seconds over BLE to get a status if it is at home. Before using the fix the issue occured after some hours and after a very short time if the vehicle was away.

If you like to get better debug information from the BLE package, you might consider to switch to rigado/ble (but it has the same problem, see my PR#76).

wimaha, 2025-01-12:
Hi, this is not the same error as #272. The fix for #272 is used.

sethterashima, 2025-01-13:
It looks like the BLE library might be encountering a packet that it doesn't know how to handle, and returns an error instead of dropping the packet.

Are you able to reproduce the issue? (This might require some other BLE that sends the unexpected packet to be nearby).

If so, could you try replacing the `return` statement in https://github.com/go-ble/ble/blob/master/linux/hci/hci.go#L436C1-L439C5 with a `continue` to see if this solves the problem?

skrul, 2025-01-14:
thanks @sethterashima this issue happens quite frequently for me. i'll try what you suggest and report back!

skrul, 2025-01-26:
@sethterashima so I wound up switching the hardware I was running this on and the problem has not reoccurred. The issue was happening with the TP Link UB500, but now that I'm using an Intel Corporation Wireless 8265 (wifi + bluetooth) and I have not seen this problem since.

closed 2025-02-21 by thefireblade