teslamotors/vehicle-command #354: Security concerns regading list-keys

Jan21493, 2025-01-19:
The "_list-keys_" command in tesla-control does not require any authentication over BLE, so it is possible to get the full key list from each vehicle nearby. I know that these keys are only public keys, but I also know that the effort to find / calculate a private key that is able to control the car is much lower if I know the public key. The distance with BLE is limited and the algorithms might be considered as safe, but in the future this may change. 

Therefore I'm asking if there are any good reasons why the "_list-keys_" command does not need any authentication over BLE?  

I don't see a problem with "_body-controller-state_" that also does not require any authentication over BLE, because if I'm only a few meters away from the car, it's easy to verify the locking state by lifting the handle. The information provided does not reveal any secrets.

sethterashima, 2025-01-20:
This isn't generally a concern; your bank and email servers send your browser their public keys every time you visit their websites, for example. But it is a good point that the command should require authentication unless there's a specific reason for it not to. I'll ask the mobile app dev team.

Jan21493, 2025-02-02:
There is a little difference: with TLS the keys are part of a certificate that has a lifetime of typically 1 or 2 years. The keys don't have to be replaced with new certificates, but they can. I'm not sure about the estimated lifetime for the keys on the vehicles, but I think it might be 10 or 20 years.

sethterashima, 2025-02-03:
A valid point, but might be worth noting that the Root CA keys typically have a much longer lifetime than leaf certificates; Google's are 20 years, for example.