teslamotors/vehicle-command #368: ClosureStatuses missing from BodyControllerState response

skrul, 2025-01-31:
It seems that `ClosureStatuses` is missing from `BodyControllerState` response. Here is the command I am using:
```
$ tesla-control -ble -key-file private.pem -vin 5YJ3E1Exxxxxxxxxx body-controller-state
{
	"vehicleLockState":  "VEHICLELOCKSTATE_UNLOCKED",
	"vehicleSleepStatus":  "VEHICLE_SLEEP_STATUS_ASLEEP",
	"userPresence":  "VEHICLE_USER_PRESENCE_NOT_PRESENT"
}
```

My charge port is currently open so I was expecting to see this in the response.

For a quick investigation, I inserted a `fmt.Println(prototext.Format(&fromVCSEC))` debug line in `vcsec::unmarshalVCSECResponse` and still only saw this:
```
vehicleStatus:  {
  vehicleSleepStatus:  VEHICLE_SLEEP_STATUS_ASLEEP
  userPresence:  VEHICLE_USER_PRESENCE_NOT_PRESENT
}
```

Is this an issue or is there an option I need to provide to return this additional data?

Lenart12, 2025-01-31:
Related https://github.com/wimaha/TeslaBleHttpProxy/issues/87

sethterashima, 2025-01-31:
It looks like when the vehicle is asleep, the charge port status will always report as closed. At least some of the other values instead will latch to their last-observed value prior to going to sleep. I don't know if this behavior will change, but it should at least be documented.

skrul, 2025-01-31:
@sethterashima thanks for the thoughts! I did some quick experimenting and I can confirm that the charge port is always reported as closed when the car is asleep (opening the frunk will force `ClosureStatuses` to be reported, including the correct state for the frunk but always a closed charge port). I also saw the "last-observed prior to going to sleep" behavior you mentioned as closing the frunk while the car is asleep does not cause its closure status to update.

I was hoping to use the charge port status from `BodyControllerState` as a way to check if the car is plugged in without waking it up. This is needed for [evcc](https://evcc.io/) since it continuously polls for charger status as part of its normal operation. Unfortunately it looks like this can not be achieved this way.