teslamotors/vehicle-command #384: command getVehicleData returns http-400

RonnyWinkler, 2025-03-13:
All commands are working except getVehicleData.

Using the proto message with protobuf.js using this JSON body doesn't work. Other commands are ok with similar body {
```
{
   "vehicleAction":{
      "getVehicleData":{
         "getChargeState":{ },
         "getClosuresState":{ }
      }
   }
}
```
The request returns http-400 error: 
command parsing error refer to https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-endpoints#vehicle-data or https://developer.tesla.com/docs/fleet-api/fleet-telemetry for vehicle data

Is the command not available yet for current firmware?
Or is there another reason what it doesn't work as expected? I am grateful for any tip :-)

zlymeda, 2025-03-21:
believe you are receiving a 400 error because you are requesting two things at once. If you use this Go repository, you can request one thing at a time. Try getting the charge state alone first, and then get the closure state.

RonnyWinkler, 2025-03-22:
> two things at once

This message is not a "oneof" type. It's a list of optional parameters like a scopes list.

ChatGPT says:

> Here is the JSON payload for a VehicleAction message that requests GetVehicleData:
> 
> ```
> {
>   "vehicleAction": {
>     "getVehicleData": {
>       "getChargeState": {},
>       "getClimateState": {},
>       "getDriveState": {},
>       "getLocationState": {},
>       "getClosuresState": {},
>       "getChargeScheduleState": {},
>       "getPreconditioningScheduleState": {},
>       "getTirePressureState": {},
>       "getMediaState": {},
>       "getMediaDetailState": {},
>       "getSoftwareUpdateState": {},
>       "getParentalControlsState": {}
>     }
>   }
> }
> ```
> 
> Explanation:
> 
>     The vehicleAction field wraps the getVehicleData message.
> 
>     The getVehicleData message requests multiple pieces of vehicle state data, with each field represented as an empty object {} to indicate the request for that data.
> 
>     If you only need specific data (e.g., just charge state and drive state), you can trim it down:
> 
> ```
> {
>   "vehicleAction": {
>     "getVehicleData": {
>       "getChargeState": {},
>       "getDriveState": {}
>     }
>   }
> }
> ```

zlymeda, 2025-03-23:
have you tried requesting a single state?
I am using this go repo, which works when requesting one thing at a time. I can read all states, but only one by one.

RonnyWinkler, 2025-03-23:
> have you tried requesting a single state?

Yes, I did. But also get a http 400 error with only one entry.

sethterashima, 2025-03-24:
This command is blocked on the backend because it provides the same functionality as the much more efficient vehicle data access methods you linked to above. It is included in the repository because it is still useful over Bluetooth in cases where Fleet API is not available or not desired.

closed 2025-03-24 by sethterashima