teslamotors/vehicle-command #390: Question: HowTo use BLE?

RonnyWinkler, 2025-04-13:
I try to get BLE connection to work in addition to FLeetAPI+signed commands
I already read that for BLE a new key pairing is necessary.

So I tried to implement the WhiteList message first using NodeJS and protobuf.js.
But I can't get it to work. Is there a kind of documentation how to use BLE including the used message types and response messages?

What I tried:
- using UnsignedMessage type incl. WhitelistOperation element.
- using ToVCSECMessage incl. submessages
For both I don't get a response message (should be FromVCSECMessage, right?)
I only get default 31 or 41byte messages where I don't know the type (and how to parse).
- How must the public key be set? Just as buffer or somehow converted? I use the existing key from FleetAPI.

I uses Tesla repo and other implementations as  example (ESPhome/PyTeslaBLE), but couldn't find a working solution.
Has anyone examples for WitelistMessage using the FLeetAPI key? Thanks in advance.

PS: I would love to get some answers or use this as a short chat, because there is no official forum to discuss such things.

sethterashima, 2025-04-14:
Protocol documentation: https://github.com/teslamotors/vehicle-command/blob/main/pkg/protocol/protocol.md.

Some notes specific to BLE (note the required two-byte length header): https://github.com/teslamotors/vehicle-command/blob/main/pkg/protocol/protocol.md#ble

Also note that keys paired with the Fleet Manager role cannot authorize commands over BLE: https://github.com/teslamotors/vehicle-command/blob/main/pkg/protocol/protocol.md#roles

It might be helpful to disable encryption on the official Golang client by switching the BLE connector to use AuthMethodHMAC [instead of AuthMethodGCM](https://github.com/teslamotors/vehicle-command/blob/058ee325e59d0290212f967f9ad94a4b61341fd5/pkg/connector/ble/ble.go#L56). Then you can run the client with `-debug` to see the protobuf messages it sends (lines with TX) and receives (lines with RX).

Feedback on the documentation is certainly appreciated.

sethterashima, 2025-04-14:
> But the documentation doesn't include details about how to use the WhitlistMessage. That's different to RoutableMessage containing signed messages.

The ToVCSEC:WhitelistOperation goes in the body (protobuf_message_as_bytes) of a RoutableMessage. Here's the Golang implementation of add-key: https://github.com/teslamotors/vehicle-command/blob/058ee325e59d0290212f967f9ad94a4b61341fd5/pkg/vehicle/security.go#L220

> Some more questions:
> 
>     * A key paired for FLeetAPI (paired with URL and hosted public key) is always role Fleet Manager?

Yes.

> 
>     * Is the error code MESSAGEFAULT_ERROR_INSUFFICIENT_PRIVILEGES the expected response when using FleetAPI key via BLE?

Yes.
 
>     * For BLE I need a second key. Is the key using the same standard (EC) like used for FleetAPI key?
> 

Yes.

>     * So using a second EC key and send it encoded as ToVCSECMessage including signedMessage/protobufMessageAsBytes/UnsignedMessage/WhitelistOperation is the right way?
> 

Yes. To enroll the second EC key, send an unsigned command that requests NFC card approval. See implementation of https://pkg.go.dev/github.com/teslamotors/vehicle-command@v0.3.3/pkg/vehicle#Vehicle.SendAddKeyRequest. You can get a feel for how this works using `tesla-control help add-key-request`.

>     * After connecting to the car via BLE, the car sends some 31byte messages. what kind of messages are these? If I send a message, then I receive such messages and a "real" answer. Can I identify the correct message or do I have to decode the bytes with a proto message type to check if that works (and it's the expected message)?

These messages are used for passive authentication (phone key). You can identify the "real" message because it will parse as a RoutableMesage and have the ToDestination field set to address you put in the corresponding request.

RonnyWinkler, 2025-04-16:
I tried to implement the WhitelistMessage based on the example:
https://github.com/teslamotors/vehicle-command/blob/82b4899b2c904b0268d78d86eb557b7220b169f8/pkg/vehicle/security.go#L320

Based on this code, the 
- WhitelistOperation is put into UnsignedMessage.SubMessage
- UnsignedMessage is put into SignedMessage.ProtobufMessageAsBytes
- SignedMessage is put into ToVCSECMessage.SignedMessage

Then ToVCSECMessage is sent via Connector (not as RouteableMessage?).
That should be this function, right? 
https://github.com/teslamotors/vehicle-command/blob/058ee325e59d0290212f967f9ad94a4b61341fd5/pkg/connector/ble/ble.go#L108

So a ToVCSECMessage is sent without RoutableMessage around?
But sending a ToVCSECMessage I don't get a response (only the 31byte messages).

Did I miss anything?

RonnyWinkler, 2025-04-19:
> Feedback on the documentation is certainly appreciated.

For me, the Whilelist process should be documented
- needed steps
- used message types and needed content
- used response message types
- typical errors, perhaps 
- if existing, differences to FleetAPI usage (signed messages after key pairing)

RonnyWinkler, 2025-04-23:
Hello @sethterashima 
happy to see a documentation is requested/coming. Thanks for bringing this on the way.

Do you have some hints atm what can be the reason for the error responses I get using the routeableMessage containing the Whitelist message (operationStatus:2 without further details)?
Do I need signature information inside the routeableMessage? Shouldn't be the case for an unsigned message, right? To check the syntax I also converted bach the bytes to message JSON to be sure, all content including the key is encoded correctly.

Thanks in advance
Ronny

RonnyWinkler, 2025-05-04:
Just a feedback:

1) Whitelist message:
I got it to work now... but not with routableMessage, but with the ToVCSECMessage instead,
But: The car doesn't send messages like "waiting for NSF" or "OK" I saw in other BLE implementations.
You have to put the NFC card on top of the console and the pair popup is shown on car screen.

2) Whitelist operation status:
For this message, the VCSEC.UnsignedMessage must be wrapped into a routeableMessage. So it's different to Whitelist. The result reports the key and slot.

3) Sending signed messages is now possible :-)

RonnyWinkler, 2025-05-10:
Hi @sethterashima 
hopefully this is the last post :-)

Finally I found the reaons why the car did not encrypt the response.
I did a wrong assumption:
I thought the flag `FLAG_ENCRYPT_RESPONSE ` is already an UINT value, becasue other languages are using UINT values for bits, like the `feature `attribute for HomeAssistant entities.

**I suggest to add a short explanation in the documentation** about this definition:
- flag value in ptoto file is a bit # starting at 0.
- The app has to do a bit shift to get the UINT value: `flags = ( 1 << universalMessageFlags.FLAG_ENCRYPT_RESPONSE )`

This can be added to: https://github.com/teslamotors/vehicle-command/blob/main/pkg/protocol/protocol.md#response-decryption
and/or as comment in `universal_message.proto`

Thanks in advance
Ronny