teslamotors/vehicle-command #393: hmac invalid errors not passed through and never cleared?

jbanyer, 2025-04-23:
This issue occurred with proxy version fef8f3a51b576385691ffc6c4d10fcdeeaed9153

We recently had a vehicle where no commands would work for many days. All commands failed with the proxy returning http 500 "context deadline exceeded". We tried rebooting the vehicle (both-scroll-wheels) but could not fix it.

Restarting the command proxy fixed the problem immediately.

In the proxy logs I noticed this:
```
2025-04-23T05:48:58Z [warn ] [52f89f8f45fe4c048905e7dc5e3c8302] Session info error: InvalidSignature: session info hmac invalid
```

Is there a problem with the handling of the "hmac invalid" situation?

sethterashima, 2025-04-25:
It looks like that situation might arise if the proxy had an incorrect public key for the vehicle.

Do you have any other logs?

Did the vehicle undergo service just before the error manifested?

iainwhyte, 2025-09-19:
Hi @sethterashima I'm seeing this trying to wake a vehicle with BLE. Yes, vehicle was just in service and has been playing up since returning. I have re-added the bluetooth key. Normal commands work, wake does not. 
`2025-09-20T08:19:12+09:30 [info ] Connected to vehicle BLE
2025-09-20T08:19:12+09:30 [info ] Connecting to car...
2025-09-20T08:19:12+09:30 [info ] Starting dispatcher service...
2025-09-20T08:19:12+09:30 [info ] Securing connection...
2025-09-20T08:19:12+09:30 [info ] Session for DOMAIN_VEHICLE_SECURITY loaded from cache
2025-09-20T08:19:12+09:30 [debug] TX: 320208023a121210cb665af3b89d4ad06ca08272982ed8ab9a03105ea5be22cde2092716aeea75b590bb5da0030252028ad16a80010a430a4104d108891019d4fa4ecf3ffbfafbdd1cfa8bb5aa2554f3721b1491de0799ca858cb633b8504f94acf5d117441bbca78c812408f5d0a5acf13552a02e6e31c3ff3a2a390a1028f46360b5ebd908d73bb98a4ca2283b120c716bd902049c41ea266912a0180e25b0e0cb002a1098a756ab7a884d6b973a043f80bba434
2025-09-20T08:19:12+09:30 [debug] RX: 1a1d12160a141885f8e1d0b6a47ea35c76207f569bd413c41b311802220101
2025-09-20T08:19:12+09:30 [warn ] [xxx] Dropping message with missing source
2025-09-20T08:19:13+09:30 [debug] RX: 32121210cb665af3b89d4ad06ca08272982ed8ab3a0208027a5c124104b572a26bcf0b332dfc3ac99c7e70d384f0cd4b0ef22f6f50ed4504bfffc6bfbfd21f28f5bc3aa6e9dea9986aaa8460861e6337098b96a68f61d6aa3de2bdeb001a109b372eb9a0031d1f390196826f2c33cd2556030000300462040802100f6a2432220a201980559006235e5406f5eb72bcd961e30e20f5efdbbf7f8ca1af236d95feb4099203105ea5be22cde2092716aeea75b590bb5d
2025-09-20T08:19:13+09:30 [warn ] [5ea5be22cde2092716aeea75b590bb5d] Session info error: InvalidSignature: session info hmac invalid
2025-09-20T08:19:13+09:30 [debug] RX: 320208003a020802521f1a1d12160a141885f8e1d0b6a47ea35c76207f569bd413c41b311802220101
2025-09-20T08:19:13+09:30 [debug] [00] Dropping message to DOMAIN_BROADCAST
2025-09-20T08:19:14+09:30 [debug] TX: 320208023a12121085fe17c9fe777ed87acec7a1926d3be29a0310eadbedcbd1c54b83f16c8e6aff4a43cca00302520269966a80010a430a4104d108891019d4fa4ecf3ffbfafbdd1cfa8bb5aa2554f3721b1491de0799ca858cb633b8504f94acf5d117441bbca78c812408f5d0a5acf13552a02e6e31c3ff3a2a390a1028f46360b5ebd908d73bb98a4ca2283b120c584adbc3f1c1ffbd1e8ea819180f25b0e0cb002a101ab2e73ba6b91fd0f7d35cee1bc4e298
2025-09-20T08:19:14+09:30 [debug] RX: 3212121085fe17c9fe777ed87acec7a1926d3be23a0208027a5c124104b572a26bcf0b332dfc3ac99c7e70d384f0cd4b0ef22f6f50ed4504bfffc6bfbfd21f28f5bc3aa6e9dea9986aaa8460861e6337098b96a68f61d6aa3de2bdeb001a109b372eb9a0031d1f390196826f2c33cd2557030000300462040802100f6a2432220a20beef07e06a7b3cd2c50e6463b50bea27ff42d4973199e655e7a623bc60bf79a2920310eadbedcbd1c54b83f16c8e6aff4a43cc
2025-09-20T08:19:14+09:30 [warn ] [eadbedcbd1c54b83f16c8e6aff4a43cc] Session info error: InvalidSignature: session info hmac invalid
`

iainwhyte, 2025-09-20:
@sethterashima removing the Tesla-control cache file fixes it. `rm /root/.tesla-cache.json` Why does the car get a new public key when it goes in for service ?

jbanyer, 2026-04-07:
Bumping this one for attention. We still encounter this issue occasionally, and it's very hard to detect because the proxy does not return a useful error to the caller.