teslamotors/vehicle-command #397: Better approach to keep car awake?

THX723, 2025-04-29:
What is the current best approach to keep the car awake without using Sentry Mode? Preferably over BLE.

The 'awake' command does what you'd expect, but does not keep the car awake even if sent periodically.

Any pointers?

sethterashima, 2025-05-06:
Normally the desired behavior is for the car to go to sleep quickly.

What problem are you trying to solve by keeping the car awake?

THX723, 2025-05-06:
> Normally the desired behavior is for the car to go to sleep quickly.
> 
> What problem are you trying to solve by keeping the car awake?

For those who run [TeslaUSB](https://github.com/marcone/teslausb), an open source Raspberry Pi project that acts as a smart USB stick to capture extended Dashcam/Sentry recordings beyond the 1 hour limit. The recordings automatically offloads (hands free) to a server when home, but takes time to complete. Currently, Sentry mode has been leveraged as a means to keep the car alive until the offload is complete. It works, but seemingly inefficient. There's got a be a better way?

sethterashima, 2025-05-13:
There's currently no elegant way to do this, since there hasn't been a need for the mobile app to keep the vehicle awake.

sethterashima, 2025-05-16:
New firmware has a setting for keeping Accessory Power on, which should obviate this issue: Controls > Charging > Keep Accessory Power On.

THX723, 2025-05-16:
> New firmware has a setting for keeping Accessory Power on, which should obviate this issue: Controls > Charging > Keep Accessory Power On.

This isn't the case for the Dashcam USB/Data port inside the glovebox. Keep Accessory Power On only applies to the 12V jack and charging only USB ports (e.g. rear console).

THX723, 2025-07-04:
> I had seen, that when asking the car repeatedly for "state charge" over BLE, then it will not go to sleep.
> Full command: tesla-control -ble {assemble_domain_string(_domain)} -session-cache ./.ble-cache.json -key-file /tmp/BLEprivatekey.pem -vin {self.vin} state charge
> 
> "body-controller-state" does not keep the vehicle awake.
> 

Interesting. Will test this out.

fabianhu, 2025-07-14:
@THX723 sorry, my bad. I just wanted to avoid having it not sleep. 
Maybe you could send some unimportant command every minute. Maybe set the charge current (looks like this even works, when not connected)
Or lock the car, even if already locked.

dzid26, 2026-11-12:
This is not accurate. Reading infotainment status every 1min keeps the car awake.  The car goes to sleep after ~90s after the last read - if the user also was not physically interacting with it in past ~10min.  

This poses an opposite problem - how to combine fast readouts with letting the car fall asleep. Currently, reading every 2min seems to be a safe minimum workaround.