teslamotors/vehicle-command #411: Remote Boombox Command Fails: "Command Not Implemented" or Tesla Protocol Error

K822843, 2025-06-23:
Hello Tesla Dev Team,

We are facing an issue with the Remote Boombox command when using this repository:

If we use the command with a signed request, we receive:`Command Not Implemented`

If we use the command without signing, we receive:

```
{
  "message": "Tesla Vehicle Command Protocol required, please refer to the documentation here: https://developer.tesla.com/docs/fleet-api/support/announcements#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning"
}
```

We are a registered third-party app using Tesla APIs, and this issue is impacting our users who rely on the remote command functionality.

Additionally, we have raised multiple Jira tickets in the Tesla developer dashboard (5 tickets pending for over a month) with no response. We kindly request timely support as this directly affects our production users.

Please assist us in resolving this issue or confirm if the Remote Boombox command is no longer supported.

Thank you.

K822843, 2025-08-06:
@patrickdemers6 @agbpatro @sethterashima

I added code in the forked repo, but I'm getting a drive pin not set error.

In the carserver.proto file, I added:

`BatchRemoveChargeSchedulesAction batchRemoveChargeSchedulesAction = 109;`

Is 109 the correct field number? I'd appreciate any suggestions you might have.

K822843, 2025-08-22:
@patrickdemers6 @agbpatro @sethterashima Anyone explain when fart command will add it's critical issue always getting error. it's big problem since last few month please review this.

```
{
  "message": "Tesla Vehicle Command Protocol required, please refer to the documentation here: https://developer.tesla.com/docs/fleet-api/support/announcements#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning"
}
```

bradleypeet, 2026-09-28:
Vox populi, vox dei, Elon. The people want that Fart 💨 command in their favorite 3rd party Tesla apps! 🙏🏻

## Summary

`remote_boombox` is documented as a Fleet API vehicle command, but there is currently no way for a third-party application to send it to a vehicle that requires the Vehicle Command Protocol:

- `tesla-http-proxy` returns `ErrCommandNotImplemented` for it, and
- sending it unsigned to the Fleet API is rejected with `403 Tesla Vehicle Command Protocol required`.

This appears to be because the public protocol has no `VehicleAction` for it, so it can't be implemented in the proxy without a protocol addition.

## Details

The proxy explicitly declines the command:

https://github.com/teslamotors/vehicle-command/blob/a4b43c1eff0e09d77deb9f2dce97031141fe8c8a/pkg/proxy/command.go#L70-L71

```go
case "remote_boombox":
    return nil, ErrCommandNotImplemented
```

`car_server.proto` has no corresponding action in `VehicleAction` (comparable commands such as `vehicleControlHonkHornAction` and `vehicleControlFlashLightsAction` are present), and "boombox" does not appear anywhere else in the repository:

https://github.com/teslamotors/vehicle-command/blob/a4b43c1eff0e09d77deb9f2dce97031141fe8c8a/pkg/protocol/protobuf/car_server.proto

The Fleet API documentation still lists the command (sound `0` = random fart, `2000` = locate ping), so applications expect it to work:
https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-commands

## Steps to reproduce

1. Pair a virtual key and send `POST /api/1/vehicles/{vin}/command/remote_boombox` with `{"sound": 0}` through `tesla-http-proxy` (latest):
   the proxy returns an error without contacting the vehicle.
2. Send the same request directly to `fleet-api.prd.na.vn.cloud.tesla.com` (unsigned):

   ```json
   {"response":null,"error":"Tesla Vehicle Command Protocol required, please refer to the documentation here: https://developer.tesla.com/docs/fleet-api/announcements#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning","error_description":""}
   ```
   HTTP 403.

Vehicle: 2026 Model Y, virtual key paired, other signed commands (e.g. `honk_horn`, `flash_lights`) work through the proxy.

## Request

Please either:

- add a `VehicleAction` for the external-speaker sound (e.g. a boombox/play-sound action with a sound ID) to `car_server.proto` and implement `remote_boombox` in the proxy, or
- if the command is intentionally unavailable to third parties, mark it as unsupported for signed-command vehicles in the Fleet API documentation, so developers don't build features around it.

Thanks!